openhwfoundation/force-riscv #2: master_run.py failed with missing handcar_cosim.so

guoheng, 2020-06-13:
Build successfully. When run master_run.py, failed with the message below:
FileNotFoundError: Path resolution for [--fpix_riscv.iss_so_path] failed, [/home/guoheng/force-riscv/utils/handcar/handcar_cosim.so] could not be located

What's the instruction of get the handcar_cosim.so?

Leo-Wang-JL, 2020-06-13:
Thank you for reporting this Issue!

The handcar_cosim.so was intended to be included in the initial commit but it was omitted unintentionally due to a .gitignore file.

The handcar_cosim.so can be also build from the code in the handcar directory.  How every we found that there are some files missing in that directory as well.  We will get that addresses ASAP so that you can confidently build the handcar_cosim.so yourself.

Leo-Wang-JL, 2020-06-14:
The missing SO file has been committed to the master branch.  Please try it out.

Leo-Wang-JL, 2020-06-14:
Submitted a pull request:
"Updated README.md to describe how to build handcar_cosim.so. Updated script in handcar/regenerate_and_build.bash that can be used to build handcar_cosim.so"

closed 2020-06-24 by guoheng