openhwfoundation/force-riscv #7: Doesn't build on CentOS 7 nor Ubuntu 20.04

tommythorn, 2020-06-25:
On CentOS Linux 7 (Core).  Cloned the repo (HEAD is at 34d8384e0a4124b5a55f6007ddcb1827462fc721).
```
$ export FORCE_CC=/usr/bin/g++
$ export FORCE_PYTHON_VER=3.6
$ export FORCE_PYTHON_LIB=/usr/lib/x86_64-linux-gnu/
$ export FORCE_PYTHON_INC=/usr/include/python3.6m
$ make
...
/usr/bin/g++ -c -Wall -std=c++11 -gdwarf-3 -m64 -Werror -O2 -Weffc++ -fvisibility=hidden -D 'ARCH_ENUM_HEADER=<EnumsRISCV.h>' -I./inc -I../base/inc -I../3rd_party/inc -I/usr/include/python3.6m -o make_area/obj/TopLevelResources.o ../base/src/TopLevelResources.cc
../base/src/TopLevelResources.cc:47:10: error: base class ‘struct option::Arg’ has a non-virtual destructor [-Werror=effc++]
   struct Arg: public option::Arg
          ^
```

On Ubuntu 20.04, same repo, but using Python 3.8:
```
/usr/bin/g++ -c -Wall -std=c++11 -gdwarf-3 -m64 -Werror -O2 -Weffc++ -fvisibility=hidden -D 'ARCH_ENUM_HEADER=<EnumsRISCV.h>' -I./inc -I../base/inc -I../3rd_party/inc -I/usr/include/python3.8/ -o make_area/obj/UtilityFunctions.o ../base/src/UtilityFunctions.cc
../base/src/UtilityFunctions.cc: In function ‘Force::uint32 Force::get_shift_amount(const string&)’:
../base/src/UtilityFunctions.cc:359:14: error: catching polymorphic type ‘class std::invalid_argument’ by value [-Werror=catch-value=]
  359 |       catch (invalid_argument) {
      |              ^~~~~~~~~~~~~~~~
../base/src/UtilityFunctions.cc:362:14: error: catching polymorphic type ‘class std::out_of_range’ by value [-Werror=catch-value=]
  362 |       catch (out_of_range) {
      |              ^~~~~~~~~~~~
cc1plus: all warnings being treated as errors
```

Leo-Wang-JL, 2020-06-26:
Thank you for reporting the issue.  What is the gcc version you are using?

tommythorn, 2020-06-26:
CentOS 7: gcc version 4.8.5 20150623 (Red Hat 4.8.5-39) (GCC)  (OK, fine that's not supported)

Ubuntu 20.04: gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2)

Leo-Wang-JL, 2020-06-26:
Thanks for providing more details.  The majority of the development currently use Ubuntu 18.04 and GCC 7.5.0.  We will provide a fix very soon, as well as having latest version of GCC in our testing process.

Leo-Wang-JL, 2020-06-29:
A pull request for the fix has been submitted last Saturday.  This issue should be addressed once the pull request is merged.

Leo-Wang-JL, 2020-06-29:
Hi Tommy,

The gcc-9 compile issue should be fixed.  Please give it a try.

Thanks,

Leo

tommythorn, 2020-06-30:
Thanks, gcc-9 issues are indeed fixed and the build makes it almost to the end:
```
... MemoryInitData.o make_area/obj/Log.o make_area/obj/FreePageRangeClaimer.o make_area/obj/InstructionSet.o make_area/obj/Operand.o make_area/obj/InstructionStructure.o make_area/obj/AddressFilteringRegulator.o make_area/obj/GenRequestQueue.o make_area/obj/PhysicalPageSplitter.o make_area/obj/PagingInfo.o make_area/obj/SchedulingStrategy.o make_area/obj/VmConstraint.o make_area/obj/GenStateTransitionAgent.o make_area/obj/OperandSolution.o make_area/obj/VmMapper.o make_area/obj/AddressSolvingShared.o make_area/obj/GenQueryResults.o make_area/obj/RegisteredSetModifier.o make_area/obj/Enums.o make_area/obj/RegisterReload.o make_area/obj/Instruction.o make_area/obj/UtilityAlgorithms.o make_area/obj/VmInfo.o make_area/obj/Random.o make_area/obj/GenInstructionAgent.o make_area/obj/Data.o make_area/obj/../utils/handcar/UopInterface.o -lpthread -static-libstdc++ -static-libgcc -L/usr/lib/x86_64-linux-gnu/ -lpython3.8m -lutil -ldl -rdynamic 
/usr/bin/ld: cannot find -lpython3.8m
$ python3 --version
Python 3.8.2
$   ls /usr/lib/x86_64-linux-gnu/libpython*
/usr/lib/x86_64-linux-gnu/libpython3.8.a@  /usr/lib/x86_64-linux-gnu/libpython3.8.so@  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1@  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
```
I didn't spend too much time, but I couldn't immediately see where the `3.8m` came from.

tommythorn, 2020-06-30:
Oh, it's hardwired in: https://github.com/openhwgroup/force-riscv/blob/master/utils/make/Makefile.common#L56
No wonder grep couldn't find any mention of "3.8m". 
I can't propose a fix as I don't understand why it's like this, but it breaks Ubuntu 20.04.
However removing the stray `m` makes the rest of the build succeed and the regression tests pass.

Leo-Wang-JL, 2020-07-01:
The 3.xm version library is faster and preferred.  But in Python 3.8 the Python library memory allocation mechanism configuration has been changed so that the 3.8m library is not produced any more.  We will have a fix soon.

Leo-Wang-JL, 2020-08-10:
The latest code should have this addressed.  Please check if it work for you now.

pdonahue-ventana, 2020-11-20:
I still get gcc-9 errors in base/src/StringUtils.cc.  Applying a similar change as https://github.com/openhwgroup/force-riscv/pull/9/commits/68108466d5d0a72fa585b18bc800fdf8f89474c7 fixes the problem.

jscheid-ventana, 2020-12-10:
> 
> 
> I still get gcc-9 errors in base/src/StringUtils.cc. Applying a similar change as [6810846](https://github.com/openhwgroup/force-riscv/commit/68108466d5d0a72fa585b18bc800fdf8f89474c7) fixes the problem.

Like this?

```
diff --git a/base/src/StringUtils.cc b/base/src/StringUtils.cc
index 71246ef..8333003 100644
--- a/base/src/StringUtils.cc
+++ b/base/src/StringUtils.cc
@@ -87,7 +87,7 @@ namespace Force {
     try {
       ret_val = stof(rStr);
     }
-    catch (invalid_argument) {
+    catch (invalid_argument&) {
       if (pErrorStatus) {
         LOG(debug) << "{parse_float} Error parsing \'" << rStr << "\', invalid argument." << endl;
         *pErrorStatus = true;
@@ -96,7 +96,7 @@ namespace Force {
         FAIL("parse-error-float-invalid");
       }
     }
-    catch (out_of_range) {
+    catch (out_of_range&) {
       if (pErrorStatus) {
         LOG(debug) << "{parse_float} Error parsing \'" << rStr << "\', invalid argument" << endl;
         *pErrorStatus = true;
```

yinhanquan, 2020-12-17:
On Ubuntu 20.04, i try gcc-7.5/gcc-9.3 , and using Python 3.8:
the error :
../base/src/TopLevelResources.cc:47:10: error: base class ‘struct option::Arg’ has a non-virtual destructor [-Werror=effc++]
   struct Arg: public option::Arg
          ^
cc1plus: all warnings being treated as errors
how can  i solve it?

noahsherrill, 2021-01-04:
@yinhanquan, I am not able to reproduce the issue you described when running Ubuntu 20.04, g++ 9.3.0 and Python 3.8.5. Could you share a list of your environment variables, i.e. the output of `printenv` and the command and arguments used to initiate the build?

noahsherrill, 2021-01-11:
@MikeOpenHWGroup, I suggest this issue be closed. All of the reported issues appear to have been resolved, except the last. I am not able to reproduce the last issue, and the reporter has not provided any additional information.

MikeOpenHWGroup, 2021-01-11:
Thanks @noahsherrill, will close the issue.

closed 2021-01-11 by MikeOpenHWGroup

tommythorn, 2021-01-11:
Sorry for note checking earlier, but FWIW it doesn't build on Ubuntu 21.04 (Hirsute Hippo):

```
/usr/include/python3.9/ceval.h:130:37: note: declared here           
  130 | Py_DEPRECATED(3.9) PyAPI_FUNC(void) PyEval_InitThreads(void);
      |                                     ^~~~~~~~~~~~~~~~~~
In file included from ../3rd_party/inc/pybind11/cast.h:21,                 
                 from ../3rd_party/inc/pybind11/attr.h:18,    
                 from ../3rd_party/inc/pybind11/pybind11.h:44,  
                 from ../base/inc/StateTransition.h:24,                                                                  
                 from ../base/src/GenStateTransitionAgent.cc:22:                                                           
../3rd_party/inc/pybind11/detail/internals.h: In function ‘pybind11::detail::internals& pybind11::detail::get_internals()’:           
../3rd_party/inc/pybind11/detail/internals.h:268:28: error: ‘void PyEval_InitThreads()’ is deprecated [-Werror=deprecated-declarations]
  268 |         PyEval_InitThreads();                                           
      |                            ^   
```

I don't have access to older distributions right at this moment.