openhwfoundation/force-riscv #13: standalone handcar share library usage?

Daichou, 2020-08-25:
I try to integrate handcar to my verification model, so I write simple test program to test how to compile handcar_cosim.so with my code. But I compile my testing code use following command:
```
g++ test.cpp force-riscv/handcar/bin/handcar_cosim.so
```
I got a lot off undefined reference:
```
force-riscv/handcar/bin/handcar_cosim.so: undefined reference to `update_vector_element'
force-riscv/handcar/bin/handcar_cosim.so: undefined reference to `update_exception_event'
force-riscv/handcar/bin/handcar_cosim.so: undefined reference to `update_generator_register'
force-riscv/handcar/bin/handcar_cosim.so: undefined reference to `f16_to_ui8(float16_t, unsigned char, bool)'
force-riscv/handcar/bin/handcar_cosim.so: undefined reference to `f16_to_ui16(float16_t, unsigned char, bool)'
force-riscv/handcar/bin/handcar_cosim.so: undefined reference to `f16_classify(float16_t)'
force-riscv/handcar/bin/handcar_cosim.so: undefined reference to `update_mmu_event'
force-riscv/handcar/bin/handcar_cosim.so: undefined reference to `find_extension(char const*)'
force-riscv/handcar/bin/handcar_cosim.so: undefined reference to `f32_to_i16(float32_t, unsigned char, bool)'
force-riscv/handcar/bin/handcar_cosim.so: undefined reference to `f16_to_i8(float16_t, unsigned char, bool)'
force-riscv/handcar/bin/handcar_cosim.so: undefined reference to `update_generator_memory'
force-riscv/handcar/bin/handcar_cosim.so: undefined reference to `f16_to_i16(float16_t, unsigned char, bool)'
collect2: error: ld returned 1 exit status
```
They look like spike function, but I compile them with spike shared library and library in handcar/standalone. I cannot eliminate those warning.
What is the correct way to use handcar api standalone?

MDBrothers, 2020-08-26:
Hi, I'm the main developer for Handcar.

Strangely, when I enter the same commands I get no such errors. What is your version of g++?

But I do have some insight for you.

**It looks like you don't have implementations for the "update_*" symbols. For examples check force-riscv/utils/handcar/SimApiHANDCAR.cc or force-riscv/handcar/tests/cosim_lest/src/cosim_lest.cc for stubs.**

**In these examples, you will notice also that the way the handcar_cosim.so is used is as a runtime loaded library. Take a closer look at force-riscv/utils/handcar/SimLoader.{h,cc}**

The other missing symbols look like they're from softfloat.h in handcar/inc or extension.h in handcar/inc.  Try with "-I<handcar_inc_dir>"?

See if the makefiles for either SimApiHANDCAR.cc or the cosim_lest examples have something that may have been missed.

Otherwise, let me look a little deeper on my side, I am not sure we use all of the half precision conversion functions as softfloat is just included piece-meal into Handcar unlike Spike; however they are declared in softfloat.h. Honestly this could be a testing hole since none of our regressions have generated instructions that require these functions.

Daichou, 2020-08-30:
I think I mistakenly used this shared library. I will re-design my test flow. Thanks your reply.

MDBrothers, 2020-08-31:
We do use this library but we load it at runtime, not link it at compile time.

Here's a small example I found. Look at the accepted answer, the "dynamic" part:
[https://stackoverflow.com/questions/1142103/how-do-i-load-a-shared-object-in-c](https://stackoverflow.com/questions/1142103/how-do-i-load-a-shared-object-in-c)

The Handcar API is officially for RV64, but it is worth a try.

I say this because the fundamental data type for the register file class is a 64bit unsigned integer no matter if Spike is operating in RV32 or RV64. This can be seen in Spike file 'decode.h'.

This has not been tested but at least in theory using the register read / write APIs should work with 32bit values so long as the correct value for the 'length' field is used and type conversion behavior within the calling context does not do anything unexpected. 

As with everything testing is important.

noahsherrill, 2021-01-04:
@Daichou, it looks like there is nothing further to be done for this issue. If that is not the case, please let me know. Otherwise, I will suggest this issue be closed.

closed 2021-01-05 by Daichou