openhwfoundation/force-riscv #15: FORCE_CC, CC, CFLAGS and LFLAGS are non-standard

jeremybennett, 2020-09-24:
The Makefiles use `FORCE_CC` and `CC` to refer to the chose C++ compiler.  By convention `CC` is used for C compiler, and `CXX` is the C++ compiler.  It would avoid user confusion to renames as `FORCE_CXX` and `CXX` throughout.

Similarly the compile time flags should be in `CXXFLAGS`, not `CFLAGS`.  Any preprocessor flags (`-I`, `-D` mostly) should be in `CPPFLAGS`.  The linker flags should be in `LDFLAGS`, rather than `LFLAGS`.

The _make_ files work fine, so there is no hard reason to change. However using standard naming may make it easier for others to pick up the project and contribute.

HTH

Leo-Wang-JL, 2020-09-25:
Good suggestions, thanks Jeremy

hanfeng0114, 2023-05-15:
Stage2 work will check the possibility to chage to cmake build system, which will solve all the related problem.
@MikeOpenHWGroup please close this issue right now.

MikeOpenHWGroup, 2023-05-16:
Thanks for the update @hanfeng0114.  Closing.

closed 2023-05-16 by MikeOpenHWGroup