openhwfoundation/force-riscv #19: generate code for BOOM core

kabylkas, 2020-11-17:
Hi :wave: 

I am trying to generate riscv code for UC Berkeley's BOOM core. 
I would really appreciate if you could give me some tips on what config files I need to edit to set force-riscv up for BOOM. For example:
* where do I change memory range?
* how can I add sections for tohost/fromhost interface?
* any other general configuration tips would be appreciated

Thank you

mtvaden, 2020-11-19:
Hello,

Thanks for using force-riscv.  To modify the memory range, edit the force-riscv/py/riscv/memory.py file (or memory32.py if you are running rv32i).
The current memory.py file has this:

def configure_memory(interface):
        interface.addMemoryRange(0, 0, 0xffffffffffff)

The first value indicates which memory space - which will always be 0 for the current RISCV ISA.
The second value is the starting address of the memory range.
The third value is the last memory address of the range.
If you want to restrict the range to 1GB starting at 0x80000000, use...
       interface.addMemoryRange(0, 0x80000000, 0xbfffffff)
You can add multiple ranges, for example...
def configure_memory(interface):
       interface.addMemoryRange(0, 0x10000000, 0x1fffffff)
       interface.addMemoryRange(0, 0x80000000, 0xbfffffff)

If you switch frequently between multiple memory configurations, create each of the desired memory.py
files (named differently).  The force-riscv/config/riscv-config file points to the memory.py file that will be used.  
(Search for "memory_file".)  You can create a riscv-config file for each configuration that you want to run.  Then
you can specify your config file as an option in the fctrl file...
          control_items = [
                     {"fname":"your_testcase_force.py",           "options":{"--cfg" : "config/your_config.config"}},
                  ]

force-riscv uses a branch-to-self to mark the end of the test case.  There is not a method to add "tohost" or "fromhost"
symbols.

noahsherrill, 2021-01-04:
@kabylkas, did you receive the answers you needed? If so, I will suggest this issue be closed.

kabylkas, 2021-01-04:
Sorry, yes this definitely helped. I will close this for now :) Thank you!

closed 2021-01-04 by kabylkas

ansyser, 2023-03-29:
You can add symbols using AddSymbol in SymbolManager.h
@kabylkas

sunshine-mh, 2025-03-13:
now do you have some methods to solve the questions about how to add tohost and fromhost symbols to end the spile?

MikeOpenHWGroup, 2025-03-13:
Hi @sunshine-mh, this issue is closed.  If you have a question or a problem, please open a new issue on this repo.