openhwfoundation/force-riscv #25: generated test examples

agrobman, 2020-12-13:
Hi, 

Could you, guys provide the generated tests *.s files for every python template?

It will greatly help to assess the tool capability...

thanks in advance

noahsherrill, 2021-01-04:
You can build and run FORCE by following the instructions here: https://github.com/openhwgroup/force-riscv#setup. Add `-k all` to the command for running the FORCE regression in order to retain the .S files for each test run, so the command to run the tests looks like `./utils/regression/master_run.py -k all`. Then, the .S files will be located in the subdirectories in `output/regression`.

noahsherrill, 2021-01-11:
@agrobman, do you have any further questions? If not, I will suggest this issue be closed.

noahsherrill, 2021-01-15:
@MikeOpenHWGroup, I suggest this issue be closed. The question was answered, and no follow-up questions were posted.

hanfeng0114, 2023-05-06:
@MikeOpenHWGroup please help to close this issue.

closed 2023-05-06 by MikeOpenHWGroup