openhwfoundation/force-riscv #26: execute_uop() implementation

ahmedgalalramzy, 2021-01-05:
Can gmp library be used for accurate computation in this function?

noahsherrill, 2021-01-11:
Hi Ramzy, I'm not very familiar with the gmp library, but based on a quick glance at the description, it would likely work. Most, if not all, of the operations currently referenced in Force code could probably be implemented using C++ operators, as long as attention was paid to making sure the operands were correctly signed or unsigned, and additional work was done to report condition flags, e.g. overflow. An important side note is that the execute_uop() interface was designed at a much earlier time in the project and is no longer being used. We don't have any plans to use it in the future, but you are, of course, free to experiment with it if you find it useful.

noahsherrill, 2021-01-15:
@ahmedgalalramzy, do you have any further questions? If not, I will suggest this issue be closed.

ahmedgalalramzy, 2021-01-16:
@noahsherrill If it is not needed in the future, then I will close the issue. Thanks

closed 2021-01-16 by ahmedgalalramzy