openhwfoundation/force-riscv #30: force riscv make error - cc1plus: error: unrecognized command line option "-std=c++11"

tvsathishriscv, 2021-03-25:
Hi Force RISCV team,

I am getting following error when I try to set up force riscv tool. Can you please help on this.

/usr/bin/g++ -Wall -std=c++11 -gdwarf-3 -m64 -Werror -O2 -Weffc++ -fvisibility=hidden -D 'ARCH_ENUM_HEADER=<EnumsRISCV.h>' -I./inc -I../base/inc -I../3rd_party/inc -I/usr/include/python3.6 -MM -MT 'make_area/obj/AddressFilteringRegulatorRISCV.o' ./src/AddressFilteringRegulatorRISCV.cc -MF make_area/dep/AddressFilteringRegulatorRISCV.d
cc1plus: error: unrecognized command line option "-std=c++11"
make[3]: *** No rule to make target `make_area/dep/AddressFilteringRegulatorRISCV.d', needed by `make_area/obj/AddressFilteringRegulatorRISCV.o'.  Stop.
make[3]: Leaving directory `/wdc/proj/riscv/scratch/sathish.venkatachalam/ehx3/force_riscv/force-riscv/riscv'
make[2]: *** [all] Error 2
make[2]: Leaving directory `/wdc/proj/riscv/scratch/sathish.venkatachalam/ehx3/force_riscv/force-riscv/riscv'
make[1]: *** [riscv] Error 2
make[1]: Leaving directory `/wdc/proj/riscv/scratch/sathish.venkatachalam/ehx3/force_riscv/force-riscv'
make: *** [all] Error 2

Clone and setup : I checked all the variable are right.

  git clone http://github.com/openhwgroup/force-riscv.git
  cd force-riscv/
   
  setenv FORCE_PYTHON_VER 3.6
  echo $FORCE_PYTHON_VER
  setenv FORCE_CC /usr/bin/g++
  echo $FORCE_CC
  setenv FORCE_PYTHON_LIB /usr/lib/x86_64-linux-gnu/
  echo $FORCE_PYTHON_LIB
  setenv FORCE_PYTHON_INC /usr/include/python3.6
  echo $FORCE_PYTHON_INC

tvsathishriscv, 2021-03-25:
in README i see that required GCC Version is 5.1 but in my environment i see only 4.4.7. Is that the issue ?

From README.md
Set the following environment variables.  FORCE-RISCV development **requires gcc 5.1** or higher and Python 3.4.1 or higher version.  The Makefile will make a guess if you don't set them, which should be suitable for modern 64-bit GNU/Linux distributions.

/usr/bin/g++ --version
**g++ (GCC) 4.4.7** 20120313 (Red Hat 4.4.7-23)

tvsathishriscv, 2021-03-25:
i tried with gcc 9.3.0

/usr/bin/g++ -Wall -std=c++0x -gdwarf-3 -m64 -Werror -O2 -Weffc++ -fvisibility=hidden -D 'ARCH_ENUM_HEADER=<EnumsRISCV.h>' -I./inc -I../base/inc -I../3rd_party/inc -I/usr/include/python3.6 -MM -MT 'make_area/obj/AddressFilteringRegulatorRISCV.o' ./src/AddressFilteringRegulatorRISCV.cc -MF make_area/dep/AddressFilteringRegulatorRISCV.d
make[2]: Leaving directory `/wdc/proj/riscv/scratch/sathish.venkatachalam/ehx3/force_riscv/force-riscv/riscv'
make[2]: Entering directory `/wdc/proj/riscv/scratch/sathish.venkatachalam/ehx3/force_riscv/force-riscv/riscv'
make[2]: Warning: File `make_area/dep/GenSequenceAgentRISCV.d' has modification time 0.0085 s in the future
make[3]: Entering directory `/wdc/proj/riscv/scratch/sathish.venkatachalam/ehx3/force_riscv/force-riscv/riscv'
make[3]: Warning: File `make_area/dep/GeneratorRISCV.d' has modification time 0.037 s in the future
make[3]: warning:  Clock skew detected.  Your build may be incomplete.
make[3]: Leaving directory `/wdc/proj/riscv/scratch/sathish.venkatachalam/ehx3/force_riscv/force-riscv/riscv'
make[3]: Entering directory `/wdc/proj/riscv/scratch/sathish.venkatachalam/ehx3/force_riscv/force-riscv/riscv'
make[3]: Warning: File `make_area/dep/EnumsRISCV.d' has modification time 0.01 s in the future
/usr/bin/g++ -c -Wall -std=c++0x -gdwarf-3 -m64 -Werror -O2 -Weffc++ -fvisibility=hidden -D 'ARCH_ENUM_HEADER=<EnumsRISCV.h>' -I./inc -I../base/inc -I../3rd_party/inc -I/usr/include/python3.6 -o make_area/obj/AddressFilteringRegulatorRISCV.o ./src/AddressFilteringRegulatorRISCV.cc
In file included from ./inc/AddressFilteringRegulatorRISCV.h:19,
                 from ./src/AddressFilteringRegulatorRISCV.cc:16:
../base/inc/AddressFilteringRegulator.h:40: error: expected ‘;’ before ‘override’
In file included from ./inc/AddressFilteringRegulatorRISCV.h:19,
                 from ./src/AddressFilteringRegulatorRISCV.cc:16:
../base/inc/AddressFilteringRegulator.h:41: error: expected ‘;’ before ‘override’
../base/inc/AddressFilteringRegulator.h:42: error: expected ‘;’ before ‘override’
../base/inc/AddressFilteringRegulator.h:44: error: expected ‘;’ before ‘AddressFilteringRegulator’
In file included from ./src/AddressFilteringRegulatorRISCV.cc:16:
./inc/AddressFilteringRegulatorRISCV.h:29: error: expected ‘;’ before ‘override’
In file included from ./src/AddressFilteringRegulatorRISCV.cc:16:
./inc/AddressFilteringRegulatorRISCV.h:30: error: expected ‘;’ before ‘override’
./inc/AddressFilteringRegulatorRISCV.h:32: error: expected ‘;’ before ‘AddressFilteringRegulatorRISCV’
./inc/AddressFilteringRegulatorRISCV.h:35: error: expected ‘;’ before ‘override’
./inc/AddressFilteringRegulatorRISCV.h:39: error: expected ‘;’ before ‘override’
./inc/AddressFilteringRegulatorRISCV.h:40: error: expected ‘;’ before ‘override’
In file included from ./src/AddressFilteringRegulatorRISCV.cc:17:
../base/inc/VmConstraint.h: In destructor ‘virtual Force::VmConstraint::~VmConstraint()’:
../base/inc/VmConstraint.h:37: error: ‘nullptr’ was not declared in this scope
../base/inc/VmConstraint.h: In constructor ‘Force::VmConstraint::VmConstraint()’:
../base/inc/VmConstraint.h:40: error: ‘nullptr’ was not declared in this scope
../base/inc/VmConstraint.h: At global scope:
../base/inc/VmConstraint.h:65: error: expected ‘;’ before ‘override’
../base/inc/VmConstraint.h:66: error: expected ‘;’ before ‘override’
../base/inc/VmConstraint.h:67: error: expected ‘;’ before ‘override’
../base/inc/VmConstraint.h:68: error: expected ‘;’ before ‘VmInConstraint’
../base/inc/VmConstraint.h:81: error: expected ‘;’ before ‘override’
../base/inc/VmConstraint.h:82: error: expected ‘;’ before ‘override’
../base/inc/VmConstraint.h:83: error: expected ‘;’ before ‘override’
../base/inc/VmConstraint.h:84: error: expected ‘;’ before ‘VmNotInConstraint’
In file included from ./src/AddressFilteringRegulatorRISCV.cc:18:
../base/inc/Constraint.h:86: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:87: error: expected ‘;’ before ‘Constraint’
../base/inc/Constraint.h:87: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:88: error: expected ‘;’ before ‘uint64’
../base/inc/Constraint.h:88: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:89: error: expected ‘;’ before ‘uint64’
../base/inc/Constraint.h:89: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:90: error: expected ‘;’ before ‘inline’
../base/inc/Constraint.h:90: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:91: error: expected ‘;’ before ‘uint64’
../base/inc/Constraint.h:91: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:92: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:93: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:94: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:95: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:96: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:97: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:98: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:100: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:105: error: expected ‘;’ before ‘void’
../base/inc/Constraint.h:105: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:106: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:107: error: expected ‘;’ before ‘override’
../base/inc/Constraint.h:108: error: expected ‘;’ before ‘override’
[... cut ...]
../base/inc/VmConstraint.h:76: note:   since type ‘Force::VmNotInConstraint’ has pure virtual functions
make[3]: *** [make_area/obj/AddressFilteringRegulatorRISCV.o] Error 1
make[3]: Leaving directory `/wdc/proj/riscv/scratch/sathish.venkatachalam/ehx3/force_riscv/force-riscv/riscv'
make[2]: *** [all] Error 2
make[2]: Leaving directory `/wdc/proj/riscv/scratch/sathish.venkatachalam/ehx3/force_riscv/force-riscv/riscv'
make[1]: *** [riscv] Error 2
make[1]: Leaving directory `/wdc/proj/riscv/scratch/sathish.venkatachalam/ehx3/force_riscv/force-riscv'
make: *** [all] Error 2

tvsathishriscv, 2021-03-26:
I got the tool working with the following setup and with some tweaks in makefile.common.

module load gcc/9.3.0
module load python/3.6.6                                                                                                                                                   setenv FORCE_PYTHON_VER 3.6.6                                                                                                                                   setenv FORCE_PYTHON_INC /wdc/apps/utilities/python-3.6.6/include/python3.6m
setenv FORCE_PYTHON_LIB /wdc/apps/utilities/python-3.6.6/lib/

MikeOpenHWGroup, 2021-03-26:
Glad to see you've made progress @tvsathishriscv.

@noahsherrill, can you comment on whether this is a valid resolution to the issue?

noahsherrill, 2021-03-26:
@tvsathishriscv, you are right that GCC 4.4.7 won't work, but GCC 9.3 should. What were the tweaks made to Makefile.common other than replacing c++11 with c++0x? That change should not have been necessary when using GCC 9.3. The second attempt looks like it may still have been referencing the GCC 4.4.7. After further modifying the environment, did you try compiling without the Makefile modifications?

noahsherrill, 2021-04-01:
@MikeOpenHWGroup, I suggest this issue be closed. The reporter appears to have resolved the issue by using a supported compiler version, and it does not look like any additional information is forthcoming.

MikeOpenHWGroup, 2021-04-01:
Agreed @noahsherrill.   In general, at OpenHW we typically give an issue reporter one calendar week to respond to the "can we close this issue" question.  You have done so, and I agree we can close it.   However, just to be fair...

Hi @tvsathishriscv, can you please provide an update to this issue?  You have not responded in one week so if you have not responded by Monday, April 5 at 08:00AM EDT, I will close it.

tvsathishriscv, 2021-04-05:
Hi All, sorry for late reply. Just hardcoded PYTHON_VER, PYTHON_LIB_TYPE in Makefile.common.
Before regression loaded glibc/2.14. 

Following steps works for me

**module load & set environment:**

module load gcc/9.3.0
module load python/3.6.6                                                                                                                                                   setenv FORCE_PYTHON_VER 3.6.6                                                                                                                                   setenv FORCE_PYTHON_INC /wdc/apps/utilities/python-3.6.6/include/python3.6m
setenv FORCE_PYTHON_LIB /wdc/apps/utilities/python-3.6.6/lib/

**Makefile difference:**

diff force-riscv/handcar/Makefile.common original/force-riscv/handcar/Makefile.common
57c57
< LFLAGS = -lpthread -static-libstdc++ -static-libgcc -L$(PYTHON_LIB) -lpython3.6m -lutil -ldl -rdynamic
---
> LFLAGS = -lpthread -static-libstdc++ -static-libgcc -L$(PYTHON_LIB) -lpython$(PYTHON_VER)$(PYTHON_LIB_TYPE) -lutil -ldl -rdynamic

% diff force-riscv/utils/make/Makefile.common original/force-riscv/utils/make/Makefile.common
64c64
< LFLAGS = -lpthread -static-libstdc++ -static-libgcc -L$(PYTHON_LIB) -lpython3.6m -lutil -ldl -rdynamic
---
> LFLAGS = -lpthread -static-libstdc++ -static-libgcc -L$(PYTHON_LIB) -lpython$(PYTHON_VER)$(PYTHON_LIB_TYPE) -lutil -ldl -rdynamic

**make stuff:**

make
make tests
**module load glibc/2.14**  

**regression run:**                                                                                                                                                  

./bin/friscv -s 0x123 -t utils/smoke/test_force.py
./bin/friscv -s 0x123 -l info -t utils/smoke/test_force.py
./bin/friscv --unknown -- --this_is_no_option

noahsherrill, 2021-04-09:
Thank you for the additional information, @tvsathishriscv. I am glad you were able to get it running. These changes look like they were all environment-related. If I am missing something, and there are issues with the FORCE-RISCV build files or code that needs to be corrected, please let me know.

MikeOpenHWGroup, 2021-04-09:
I agree with @noahsherrill, this looks like its a env set-up issue, not a FORCE-RISCV issue per-ce.  Therefore, I will close this issue.

@tvsathishriscv, if you believe there is additional information we could add to the README, please open an issue against that document.

Thanks to both of you for grounding this out.

closed 2021-04-09 by MikeOpenHWGroup