openhwfoundation/force-riscv #31: new rori instr, that added by myself refer to force_new_instr_toturial.pdf, can not run by fpix and handcar in force instr generating.

MappleM, 2021-07-28:
Hello, I try to add a new instr rori followed toturial step by step from doc/force_new_instr_toturial.pdf. But I have a problem. Now, rori instr can be generated by force correctly after I do all steps on pdf file. when I try to run ELF file using fpix/handcar, there is a error in fpix_sim.log file. It seems that modificaton to handcar have not take effect. I have added rori.cc in src, added rori.h in inc/insns, modified encoding.h file.
firstly, I added src/rori.cc file.
[image]
then, I added inc/insns/rori.h.
[image]
I added some contents about RORI_MASK, RORI_MATCH and DECLEAR_INSN into encoding.h file.
[image]
[image]
then I remake handcar_cosim.so and copy it to utils/handcar/
then, I added new instr into force instr-start-xml file and remake forece and fpix.
I wrote a simple template to test new rori instr.
[image]
as it shows in test_force.S file, rori instr was generated and its format is correctly.
[image]
 as show in fpix_sim.log file, rori instr can not be decode correctly, it can not be identified and trigger a illigeal instr exception.
[image]
tipis: rori (args unknown) 
I dont known why it error, could anyone help me? 
Please.

noahsherrill, 2021-07-29:
I do not know the exact cause of or remedy for these issues. However, I have a couple of suggestions to try. You may need to add a line in inc/insn_list.h of the form `DEFINE_INSN(rori)`. Concerning the "args unknown" issue, I think if you add a line of the form `DEFINE_ITYPE(rori)` to src/disasm.cc, it will likely resolve it.

MappleM, 2021-08-02:
Yes，It runs right after your guide! 
Thanks.

MikeOpenHWGroup, 2021-08-02:
Hi @MappleM, if this problem has been resolved to your satisfaction, please close this issue.   Thanks.

closed 2021-11-12 by MikeOpenHWGroup