openhwfoundation/force-riscv #33: Generate code for only non-privileged ISA

seferbey, 2021-08-23:
Hi,

I am trying to generate riscv code for IMAFD extension. The core has only machine mode. Is it possible?

I would really appreciate if you could give me some tips on what would be necessary changes (If possible of course). 

From the config file, I removed <instruction_files> of c, v, priv and zfh instructions.
I get -->
[fail]Instruction with ID "MRET##RISCV" not found.

Thanks in advance.

noahsherrill, 2021-08-30:
@seferbey, did this help to resolve the error you encountered? Do you have any further questions?

seferbey, 2021-08-31:
Error is caused by the removal of priv_instructions.xml  from the config file. I changed "RET" weight to 0. But, it still generates SRET and MRET instructions. (Maybe for traps, I don't know). Thank you for the response anyway.

MikeOpenHWGroup, 2021-08-31:
Hi @seferbey, to me it does not appear as if the issue has been resolved.  Can you share the config file you are using?  (Attached it to this issue.)

seferbey, 2021-08-31:
Here is the config file I use. If I remove "priv_instructions.xml", I get an error. (changed extension to drag&drop)

[riscv_rv64.config.txt](https://github.com/openhwgroup/force-riscv/files/7084036/riscv_rv64.config.txt)

Here is the suggested weight change. (changed extension to drag&drop)

[general_choices.xml.txt](https://github.com/openhwgroup/force-riscv/files/7084053/general_choices.xml.txt)

With this configuration, I still get "SRET" and "MRET" in the generated assembly file.

noahsherrill, 2021-09-07:
Yes, you are right that the exception handlers will still be generated with MRET and SRET instructions. You can pass the option `--options NoHandler=1` to FORCE to skip exception handler generation. In a fctrl file, this would like something like `{"fname": "<test_file_path>", "generator": {"--options": '"NoHandler=1"'}}`.

noahsherrill, 2021-09-13:
@seferbey, did this remove the remaining SRET and MRET instructions? Do you have any further questions?

seferbey, 2021-09-13:
Yes, it resolved the issue. Thank you.

closed 2021-09-13 by seferbey