openhwfoundation/force-riscv #34: Front-end error when using getConditionalBranchToAddress(): 'AssemblyHelperRISCV' object has no attribute 'error'

Wlgen, 2021-11-03:
I am developing a template which generates code with loops and a series of branches and I am using the `genConditionalBranchToAdress()` method to generate a branch instruction to return to the beginning of the loop and keep iterating. The error comes when setting a non-valid address and has to return an error message, then the python interpreter returns the following error:
```
[fail]Front-end error reported: AttributeError: 'AssemblyHelperRISCV' object has no attribute 'error'
  File "/home/ivan/Repos/BSC/force-riscv/py/base/GenThreadExecutor.py", line 62, in executeGenThread
    aGenThread.generate()
  File "/home/ivan/Repos/BSC/force-riscv/py/base/GenThread.py", line 148, in generate
    seq.run()
  File "/home/ivan/Repos/BSC/force-riscv/py/base/Sequence.py", line 49, in run
    self.generate(**kargs)
  File "/home/ivan/Repos/BSC/force-riscv/BSC/BSC_self_check_test.py", line 51, in generate
    self.genScalarEquivalentLoopVV(vd, vs2, vs1, ins)
  File "/home/ivan/Repos/BSC/force-riscv/BSC/BSC_self_check_test.py", line 165, in genScalarEquivalentLoopVV
    r_index, vl, loop_start_address, "NE",
  File "/home/ivan/Repos/BSC/force-riscv/py/riscv/AssemblyHelperRISCV.py", line 397, in genConditionalBranchToAddress
    br_offset = self.getBranchOffset(aTargetAddr, 12)
  File "/home/ivan/Repos/BSC/force-riscv/py/base/AssemblyHelper.py", line 337, in getBranchOffset
    self.error(

[FAIL]{front-end-error-reported} in file '../base/inc/py_modules/PyLog.h' line 59 func 'operator()'.
```
I managed to solve the issue by importing `Log` at the beginning of the `py/base/AssemblyHelper.py` file but I wanted to report this because there may be a better fix to this that I am not aware of.

noahsherrill, 2021-11-05:
Thank you for the report. It should have originally been written as `self.mSequence.error(` on Line 337 instead of `self.error(`. However, `Sequence.error(` is now deprecated in favor of `Log.error(`, so your fix is a good one. If you are inclined, please submit a pull request with your fix. Otherwise, we can address it in our next update.

Wlgen, 2021-11-09:
Thanks for the reply. I will create a pull request with the fix then.

noahsherrill, 2021-11-16:
@MikeOpenHWGroup, I believe this issue should be closed, as Wlgen has submitted a fix.

closed 2021-11-16 by MikeOpenHWGroup