openhwfoundation/force-riscv #36: Instruction address issues when generating branch instructions

Wlgen, 2021-11-12:
Hello, I am trying to create a template for a test that generates a loop and I have encountered a few problems when trying to generate branch instructions.
This is the piece of code that generates the problems:
```python
# Check result from both registers and branch if result mismatch
assembly_helper.genConditionalBranchToLabel(
     r_vd, r_res, 4, "NE", "error_code"
)
# Add one to the index register
record_id = self.genInstruction("ADDI##RISCV",
                   {
                        "rd": r_index,
                        "rs1": r_index,
                        "simm12": 1})
# Check if end of loop
assembly_helper.genConditionalBranchToAddress(
    r_index, vl, loop_start_address, "NE",
)
# Set ok code to mem region
record_id = self.genInstruction("SW##RISCV",
                   {
                       "LSTarget": 0x80808080,
                       "LSData": 0xBEEFBEEF
                   })
# Jump to the end of the code
assembly_helper.genRelativeBranchToLabel(4, "end_code")
Log.notice("Branch records")
Log.notice(str(assembly_helper._mBranchRecords))
assembly_helper.addLabel("error_code")
```
The issue I am having with this is that when force generates the first BNE instruction then it reserves some contiguous memory addresses and when its time to generate the ADDI instruction it detects that the address at which that ADDI should go is reserved and jumps to another address. Then I get a couple of weird things, first of all that, since there has been an unexpected jump to another address the offset for the second branch is too large, but the instruction is generated with an offset of 0. This branch instruction is generated using the `genConditionalBranchToAddress` method, in which I found an error in a previous issue I posted, and passing as target address the PC of the first instruction of the loop which is stored in `loop_start_address`. Then the next instructions are generated correctly until the `error_code` label is set, then when force goes to update the branch instruction's offset it fails with the following errors:
```
[notice]Front-end: INTERNAL ERROR (EquivalentLoopVV) BRANCH TARGET (error_code) MISMATCH:
[notice]Front-end: INTERNAL ERROR (EquivalentLoopVV) BRANCH ADDRESS: 0x8011101c, BRANCH OFFSET (# OF INSTRS): 0, LABEL OFFSET: -131014
```
This is the log where you can see the reservation of memory when generating the branch instruction and then the generation of the ADDI instruction in a bigger address and in the end the generation of the second branch address with offset 0.
```
[notice]Generating: BNE##RISCV
[info]constraint already forced, ignore reservation check
[info]constraint already forced, ignore reservation check
[info]{PageRequestRegulator::RegulateBranchPageRequest} exception type: InstructionPageFault exception name: InstructionPageFault exception constraint type: Invalid
[info] opname=rs2
[info] opname=rs1
[info] opname=Branch-simm12
[notice]Operand : Branch-simm12 mask 0x0 value 0x0
[notice]{OperandGenerate} Operand : simm12 Constraint: 0x0 value 0x0
[notice]{FullsizeConditionalBranchOperandConstraint::SetConditionalBranchTaken} condition branch is set to taken
[info]{PhysicalPageSplitter::GetPhysicalPageSplit} page crossing, VA=0x8011101c=>part 1 PA [0]0x8011101c size=4, part 2 PA [3370147293]0x0 size=0
[notice]Committing instruction "BNE x2, x13, 0" at 0x8011101c=>[0]0x8011101c (0xd11063) gen(0)
[info]InitializeMemory: MemoryInitRecord: address=0x000000008011101c mem-ID=0 size=4 element-size=4 type=Instruction data=6310d100
[trace]{MemoryBank::InitializeMemory} bank=Default start_addr=0x8011101c end_addr=0x8011101f
[info]current source entropy:8, resource type: 0
[notice]{MemoryManager::ReserveMemory} MemoryReservation: BNT141 [Default] Initial: 0x8011101c-0x8011103f
[notice]Front-end: {'error_code': [(2148601884, 0)]}
[notice]{Generator::ProcessGenRequest} instruction space not free at 0x80111020 pending request: GenInstructionRequest: ADDI##RISCV
[notice]{GenSequenceAgent::HandleRequest} HandleRequest is called with : EscapeCollision
[info]{GenSequenceAgentRISCV::GetShortBranchConstraint} PC-relative-branch PC offset constraint: 0x80011020-0x80211041
[info]{PageRequestRegulator::RegulateBranchPageRequest} exception type: InstructionPageFault exception name: InstructionPageFault exception constraint type: Invalid
[notice]{GenSequenceAgent::EscapeCollision} space left at pc 0x80111020 intersection start at 0x80111040 size 0x3b087f8 use short branch? 1 target address: 0x80130878
[notice]Generator mode changed to: NoIssNoEscape from: NoIss
[notice]{GenSequenceAgent::HandleRequest} HandleRequest is called with : BranchToTarget
[notice]Generating: JAL##RISCV
[info]constraint already forced, ignore reservation check
[info]{PageRequestRegulator::RegulateBranchPageRequest} exception type: InstructionPageFault exception name: InstructionPageFault exception constraint type: Invalid
[info] opname=rd
[info] opname=Branch-simm20
[notice]Operand : Branch-simm20 mask 0x0 value 0x0
[notice]{OperandGenerate} Operand : simm20 Constraint: 0xfc2c value 0xfc2c
[info]{PhysicalPageSplitter::GetPhysicalPageSplit} page crossing, VA=0x80130878=>part 1 PA [0]0x80130878 size=4, part 2 PA [3370147293]0x0 size=0
[notice]Committing instruction "JAL x0, 64556" at 0x80111020=>[0]0x80111020 (0x591f06f) gen(0)
[info]InitializeMemory: MemoryInitRecord: address=0x0000000080111020 mem-ID=0 size=4 element-size=4 type=Instruction data=6ff09105
[trace]{MemoryBank::InitializeMemory} bank=Default start_addr=0x80111020 end_addr=0x80111023
[notice]retire dest stage: e, access: 0x7, size: 1, type: 0
[info]current dest entropy:2, resource type: 0
[notice]Generator mode changed to: NoIss from: NoIssNoEscape
[notice]Generating: ADDI##RISCV
[info]constraint already forced, ignore reservation check
[info]constraint already forced, ignore reservation check
[info] opname=simm12
[notice]{OperandGenerate} Operand : simm12 Constraint: 0x1 value 0x1
[info] opname=rs1
[info] opname=rd
[notice]Committing instruction "ADDI x25, x25, 1" at 0x80130878=>[0]0x80130878 (0x1c8c93) gen(0)
[info]InitializeMemory: MemoryInitRecord: address=0x0000000080130878 mem-ID=0 size=4 element-size=4 type=Instruction data=938c1c00
[trace]{MemoryBank::InitializeMemory} bank=Default start_addr=0x80130878 end_addr=0x8013087b
[info]current source entropy:8, resource type: 0
[info]current dest entropy:3, resource type: 0
[notice]GenQueryAgent::HandleQuery : GenState
[info]{GenQueryAgent::GenState} state: PC=0x8013087c
[notice]GenQueryAgent::HandleQuery : BranchOffset
[notice]Generating: BNE##RISCV
[info]constraint already forced, ignore reservation check
[info]constraint already forced, ignore reservation check
[info]{PageRequestRegulator::RegulateBranchPageRequest} exception type: InstructionPageFault exception name: InstructionPageFault exception constraint type: Invalid
[info] opname=rs2
[info] opname=rs1
[info] opname=Branch-simm12
[notice]Operand : Branch-simm12 mask 0x0 value 0x0
[notice]{OperandGenerate} Operand : simm12 Constraint: 0x0 value 0x0
[notice]{FullsizeConditionalBranchOperandConstraint::SetConditionalBranchTaken} condition branch is set to not taken
[info]{PhysicalPageSplitter::GetPhysicalPageSplit} page crossing, VA=0x8013087c=>part 1 PA [0]0x8013087c size=4, part 2 PA [3370147293]0x0 size=0
[notice]Committing instruction "BNE x25, x23, 0" at 0x8013087c=>[0]0x8013087c (0x17c9063) gen(0)
```
I'd like to know if this behavior is intended and I am not understanding something or if there is some kind of problem with it.
Thanks in advance.

noahsherrill, 2021-11-12:
Thank you for the report. The memory block that gets reserved following the BNE instruction is for branch not-taken instruction generation. In most cases, this is desirable to ensure that FORCE can generate the branch not-taken instructions to facilitate simulation of various branch predication and speculative execution strategies; however, in tests that are highly directive, this mechanism can be undesirable. It can be disabled via the NoBnt option to the genInstruction() method, but AssemblyHelper isn't currently set up to facilitate this. I need to spend some time to look into this further.

Wlgen, 2021-11-15:
Thanks for the answer, now I understand a few more things. But knowing about this branch not-taken code, is there a way to tell force-riscv what code to put inside this kind of code? Because right now I'm making the loop code practically by hand but I was wondering if there were any tools, like genConditonalBranch for the branches, that helped in loop generation specifically.

noahsherrill, 2021-11-16:
There are two features that may be of interest to you. The first is the `Sequence.setBntHook()` method. `setBntHook()` allows you to specify a method to be invoked to generate the instructions along the not-taken path. You can look at `tests/riscv/bnt/speculative_bnt_force.py` for an example of how to use it.

The feature I think would probably be more useful to you is the `LoopControl` class in the `riscv.Utils` module. It automatically sets a loop counter and generates the necessary branch and jump instructions to return control to the start of the loop. The `tests/riscv/loop` directory has several templates that use this feature.

Wlgen, 2021-11-18:
Both of these are really useful! I think I will be able to manage better using these, thanks a lot. Those were all the questions I had so I think that this issue can be closed.