openhwfoundation/force-riscv #37: Adding pre-compiled C code to the binaries generated by force-riscv

Wlgen, 2021-12-02:
Hello, I need to generate a set of tests that use a set of methods that were created in C code and I was wondering if there is a way to insert this code, be it with the C code itself or the compiled binary, so that force is able to generate these pieces of code within the binaries that it generates. I was not able to find any mechanism that let me do this so I wanted to ask before giving up.
Thank you.

noahsherrill, 2021-12-02:
Unfortunately, FORCE doesn't have any mechanisms to support this. FORCE outputs an ELF file with the instructions and data necessary for the generated test, but there isn't any way that I know of that one could link that ELF file to call into another binary.

Wlgen, 2021-12-03:
And what about assembly code? Is there a way to insert a piece of assembly riscv code instead of the binary?

Wlgen, 2021-12-16:
I see, then that solves my doubts, thank you very much for the reply.

closed 2021-12-16 by Wlgen