openhwfoundation/force-riscv #39: How to generate correct code for Machine-mode only supported core?

MasterJerryZh, 2022-01-09:
Hi, I am using  force-riscv to verifi my cpu core in my dv env. But there are some mismatch code in generated elf file for example that csrrw satp/fcsr/stvec/scause are here. My cpu core is a Machine-mode only riscv core, and it not support U-mode or S-mode. These U/S-mode csr are not exist in my core and they will be decode as illegal instruction exception. I use this cmd "/opt64/MySoftware/force-riscv/bin/friscv -t um_itree_01_force.py" to generate elf file. Force will use default config file load from force-riscv/config/riscv_rv64.config  because I give not a valid param --cfg to it. I want know how should I do to generate correct code without any S/U-mode csr init code in the beginning for my M-mode only riscv core ? Thanks.

noahsherrill, 2022-01-13:
Unfortunately, we don't have support for generating for an M-mode only core at this time. If you are willing to modify the source code, you could try modifying `riscv/src/InitialSetupRISCV.cc` to remove references to the registers not supported by the M-mode only core. Lines such as `mpGenerator->InitializeRegisterFields("satp", field_map)` would need to be removed to avoid initializing these registers.

You might also need to make sure the register_file tags in the config file being used only reference files with registers that are legal for your design, but that may not be strictly necessary.

MasterJerryZh, 2022-01-16:
Thanks. I will try to midify it according your guiding.

closed 2022-01-16 by MasterJerryZh