openhwfoundation/force-riscv #40: How to call OvpSimPlus in master_run.py instead of fpix

CodeDance364, 2022-01-19:
Hi guys
As the force-riscv README.md said,we can run the generation with master_run.py,just as utils/regression/master_run.py -f examples/riscv/_def_fctrl.py -c utils/regression/config/_riscv_rv64_fcfg.py -k all.But we found it will call fpix as ISS to execute the ELF that generated before,and the execution showd FAIL,single run the ISS cmd with no response,seems timeout.

Anyone could help to confirm whether master_run.py could use OVPsimplus to run the ELF just as the way of ../bin/friscv,instead of using fpix,since OVPsimplus has a better behavior on coverage than fpix.

Thanks.

noahsherrill, 2022-01-20:
Unfortunately, we don't presently have any support for executing OVPsimPlus using master_run.py. Imperas previously contributed some scripts intended to enable getting functional coverage using OVPsim. You can find them here: https://github.com/openhwgroup/force-riscv/blob/master/target. If you have any questions about how to use them, you will probably need to contact Imperas.

The failure when running Fpix is unexpected. If you would like some assistance in figuring out why that is failing, I can attempt to help. Any console error messages, the contents of the gen.err file and the contents of the forrest.log file would be useful in trying to determine the cause of the failure.

noahsherrill, 2022-02-01:
@CodeDance364, do you have any further questions? If not, I will suggest this issue be closed.

hanfeng0114, 2023-05-15:
@CodeDance364 It seems there has been long time no response for this issue. Pleaes issue another issue if there is more question. 
@MikeOpenHWGroup please close this issue right now.

closed 2023-05-15 by MikeOpenHWGroup