openhwfoundation/force-riscv #46: Removing support for certain extensions

snipppy, 2022-02-07:
Hi there,

I'm attempting to use the pre-existing tests for a simpler, open source core that doesn't have several of the extensions listed here such as vector, floating point, etc. My understanding is that in order for this to happen, I'll need to:

1. Edit/create a new configuration file with desired existing tests
2. Modify tests themselves and remove content so they don't reference extensions the targeted core doesn't support
3. Edit platform specific information such as memory.py for initializing the core

The user manual has been very useful so far! It might be useful to create a short note in the README for editing source code directly for those looking existing tests, as I've noticed a couple other issues that ask similar questions (e.g. machine-mode only tests). Are there plans to making supported extensions parameterizable in the regression test?

Thanks!

noahsherrill, 2022-02-22:
Thank you for the feedback and question. We have had one or two discussions about enhancing the extension configurability. I agree that there is significant room for improvement in ease-of-use in this area. Unfortunately, we don't have anyone to work on it at the moment.

I would generally prefer not to add anything to the README that involves editing the source code, as I think that is more detailed information that most users need or want. However, we have a fairly sparse Wiki at the moment, and some tips on editing the source code to configure FORCE-RISCV could be useful in that area.

One other note concerning the pre-existing test templates: you are, of course, welcome to use them as you see fit. However, please be aware that the test templates that come with FORCE-RISCV were mostly written for the purpose of verifying the features of the tool itself. They were not meant as a suite of tests to verify any hardware design. We expect users will generally want to write their own test templates for that purpose.

MikeOpenHWGroup, 2022-02-23:
Hi @snipppy.  Would you consider submitting a pull-request with updates to either the generator or documentation?  This is how open-source works.  :wink:

hanfeng0114, 2023-05-15:
@snipppy Can we close this issue now?

hanfeng0114, 2023-10-21:
close for no reponse.

closed 2023-10-21 by hanfeng0114