openhwfoundation/force-riscv #52: how to generate codes without any physical registers

erdevonmatteo11, 2022-04-21:
Hi,

I am trying to generate codes without any physical registers. Is it possible? And would you mind telling me which script files to modify & how to modify?

Thank you in advance.

erdevonmatteo11, 2022-04-21:
Hello @MikeOpenHWGroup , I mean the "physical registers" are such as mhartid, misa, mstatus and so on, which are classified as physical registers in force-riscv scripts as I notice.

In the assembly files generated by force-riscv, they are written just like:
CSRRS x27, mhartid, x0
CSRRW x0, misa, x21
CSRRW x0, mstatus, x20
CSRRW x0, satp, x31
CSRRW x0, fcsr, x15

Well, I don't know how to make force-riscv not to generate codes like those above. It seems force-riscv doesn't have proper command flags to tackle with this. I tried to modify some scripts but found there may be a lot lines to be commented or deleted.

noahsherrill, 2022-04-25:
By default, FORCE attempts to configure the CSRs for successful execution. However, all of the register initialization instructions can be bypassed by including `"generator": {"--options": '"SkipBootCode=1"'}` in the `_fctrl.py` file. An example of this can be found in `tests/riscv/APIs/_def_fctrl.py`.

erdevonmatteo11, 2022-04-26:
> 

Thank u for your reply.

I tried _tests/riscv/APIs/_def_fctrl.py_ as you mentioned. However, I found the generated _skip_boot_force.Default.S_ still has a lot of CSRs sentences. Is there a bug inside the present force-riscv codes?

[image]

noahsherrill, 2022-04-26:
You are right that FORCE also generates the exception handler code, which accesses CSRs, although it will not be executed during simulation unless an exception is triggered. To remove that, augment the options such that `"SkipBootCode=1"` becomes `"SkipBootCode=1,NoHandler=1"`.

There is one other place where an access to `mhartid` is generated. I don't know of any existing option to bypass this. However, you can accomplish it by modifying `py/riscv/ThreadSplitterSequence.py`. Inserting a `return` statement at the top of the `ThreadSplitterSequence.generate()` method will probably be sufficient. It will likely break any test that does not set the `SkipBootCode` option.

erdevonmatteo11, 2022-04-27:
@[noahsherrill](https://github.com/noahsherrill) Thank u so much. Following on ur reply, I managed to solve it and was able to locate related codes.

MikeOpenHWGroup, 2022-04-27:
Hi @erdevonmatteo11, if your issue is resolved, please close this issue.   Thanks!

closed 2022-04-28 by erdevonmatteo11