openhwfoundation/force-riscv #53: error found while convert generated elf to binary

erdevonmatteo11, 2022-04-22:
Hi,

I try to convert the ELF file generated by force-riscv to binary by **riscv64-linux-gnu-objcopy**. However, I find that the result binary file has a very large size more than 800 MB. In comparison, the ELF file created by force-riscv is only 1.6 kB. What is wrong inside?

Below is the screen shot:
[image]

MikeOpenHWGroup, 2022-04-22:
This looks like a toolchain, not force-riscv issue.  Maybe have a look at your crt0.S and/or link.ld.

erdevonmatteo11, 2022-04-22:
> This looks like a toolchain, not force-riscv issue. Maybe have a look at your crt0.S and/or link.ld.

the riscv toolchain I use is officially released on ubuntu and installed by apt install command, not compiled individually. Therefore, I think my crt0.S, lind.ld and related files could be no problem.

MikeOpenHWGroup, 2022-04-22:
The crt0.S, link.ld, etc. are specific to your environment.   A good discussion of this can be found [here](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40p/bsp/README.md).

noahsherrill, 2022-04-25:
I would note that the ELF files that FORCE generates only contain the values necessary to initialize data and instruction memory in a bare-metal environment. They are very lightweight and do not contain many of the additional elements that one would find in a binary designed for execution in an OS environment.

noahsherrill, 2022-05-02:
@erdevonmatteo11, do you have any further questions or comments concerning this issue? If not, I will suggest this issue be closed.

erdevonmatteo11, 2022-05-02:
thank u @noahsherrill. I have no more question about it.

closed 2022-05-02 by erdevonmatteo11

xusq123, 2024-02-23:
@erdevonmatteo11 have  you know the mean of each line in .S file  I have translate it in .obj file  and has 8 text   but i dont know the mean of each text