openhwfoundation/force-riscv #55: bug may exist in page-fault exception handler

niuqiang-real, 2022-04-25:
Hi there,

     I'm currently using FORCE to generate some load/store instructions without preamble instructions. Lots of my generation failed for one same reason. After some debugging, I believe these fails were caused by a bug in FORCE. And I would be grateful if you can help me to confirm it.
 
     In the official description about Sv48, there is a description about one of there exceptions:  
> Instruction fetch addresses and load and store effective addresses, which are 64 bits, must have bits 63–48 all equal to bit 47, or else a page-fault exception will occur. 

     Comprehensive page fault exception handler uses an OS like approach to service many page related exceptions. It checks the faulting address, tries to correct it when it's illegal, writes the modified address into XTVAL and returns.  
     BUT this is not enough. After returning from exception subroutine, PC points to the same instruction that caused this page fault exception. Executing it will raise the same exception again. Core will repeat this process until exceed instruction upper-limit and generation fails.

Thanks!

noahsherrill, 2022-04-25:
Thank you for the detailed analysis. I agree with your assessment; this is an issue. Most of the page fault handler logic is intended to correct an error in the page tables and re-execute the instruction. However, in the case, I can't think of a good mechanism by which we could successfully re-execute the instruction because the fault is with the instruction itself. I think we will likely have to adjust the handler to skip the instruction when it detects the invalid target address.

niuqiang-real, 2022-04-26:
> Thank you for the detailed analysis. I agree with your assessment; this is an issue. Most of the page fault handler logic is intended to correct an error in the page tables and re-execute the [...]

niuqiang-real, 2022-04-26:
> Thank you for the detailed analysis. I agree with your assessment; this is an issue. Most of the page fault handler logic is intended to correct an error in the page tables and re-execute the [...]

Thanks for your reply.

niuqiang-real, 2022-04-26:
> Thank you for the detailed analysis. I agree with your assessment; this is an issue. Most of the page fault handler logic is intended to correct an error in the page tables and re-execute the [...]

Thanks for your replay! Skipping the instruction when it detects the invalid target address will solve this problem. But generating load/store instructions without preamble instructions will raise too many exceptions, which is not what I want. I want to test register dependency with group of load/store instructions. Can FORCE read registers from handcar and automatically select a suitable one  for the instruction?

noahsherrill, 2022-04-26:
Yes, to an extent. FORCE does read registers from Handcar to maintain an accurate view of the simulation state. By default, FORCE chooses register operands that yield valid target addresses when explicit register operands are not passed to the `genInstruction()` method. FORCE can also be configured to attempt to generate dependencies. However, the dependency generation isn't perfect; it can't always yield the desired dependencies.

Most of the relevant options are in `riscv/arch_data/general/dependence_choices.xml`. You can adjust the options there to attempt to yield the types of dependencies you are interested in. The other option that might be helpful is in `riscv/arch_data/general/operand_choices.xml`. Setting 100% weight to `Load store address` for the `Load data` choices helps the dependency generation by having the destination registers of load instructions receive values from memory that are valid addresses.

noahsherrill, 2022-04-28:
@niuqiang-real, I am going to submit a fix for the page fault handler soon. I opened a separate issue (#56) for the register dependency generation in case further discussion and/or action is required.

PR #57 (2022-04-29): Skip instructions with invalid target addresses
Fixes #55
closed 2022-04-29 by MikeOpenHWGroup