openhwfoundation/force-riscv #56: Test register dependency with load/store instructions

noahsherrill, 2022-04-28:
I want to test register dependency with a group of load/store instructions. Can FORCE read registers from Handcar and automatically select a suitable one for the instruction?

noahsherrill, 2022-04-28:
Yes, to an extent. FORCE does read registers from Handcar to maintain an accurate view of the simulation state. By default, FORCE chooses register operands that yield valid target addresses when explicit register operands are not passed to the `genInstruction()` method. FORCE can also be configured to attempt to generate dependencies. However, the dependency generation isn't perfect; it can't always yield the desired dependencies.

Most of the relevant options are in `riscv/arch_data/general/dependence_choices.xml`. You can adjust the options there to attempt to yield the types of dependencies you are interested in. The other option that might be helpful is in `riscv/arch_data/general/operand_choices.xml`. Setting 100% weight to `Load store address` for the `Load data` choices helps the dependency generation by having the destination registers of load instructions receive values from memory that are valid addresses.

niuqiang-real, 2022-05-05:
> 
Thanks for open this issue to discuss dynamic register selection. I think this is one of the most important advantages of dynamic generation. And I'm very interested in the register selection scheme in FORCE~

I found two kinds of register selections when generating target address without preamble sequences:
1. Select uninitialized/free registers. Calculate the base address after target VA is determined and write it into the selected register as initial value.
2. Select initialized registers. Read values contained in these registers and calculate select the register with value satisfies constraints.
Is  there any other kind of register selection?

niuqiang-real, 2022-05-06:
Thank you for your supplement~
You mentioned that FORCE may reserve register by itself. If a register is reserved without user's knowledge,  some interesting scenarios may not be able to generate. So why would FORCE do that?

noahsherrill, 2022-05-06:
One case is when using "fast" exception handlers. In order to execute a minimal amount of instructions before returning from a "fast" handler, the handler is permitted to modify register state without restoring it. The comprehensive handlers do not do this; they restore all register values before returning, which requires a significant amount of additional instructions. FORCE reserves 2 registers for the "fast" handlers to use at will. The purpose of reserving them is to discourage the test template generation from selecting those registers because having their values modified by the exception handlers could yield undesirable results.

Another case is the address table. FORCE maintains a pointer into a table with valid instruction addresses. The exception handlers use this table to jump to a known good location when other forms of recovery aren't feasible, such as an unrecoverable instruction page fault. Modifying this register would break the recovery mechanism.

Another scenario in which FORCE reserves registers is when the value of a register is nondeterministic. One common example is with `SC` instructions, which write either a 0 or 1 to the destination register. The same system could execute the same test with a different return value from `SC`.

FORCE has a special type of reservation for such cases, which marks the register as "Unpredictable". This only prevents reading from the register; it is still allowed to be written to. This avoids many frustrating cases in which the system under test could become out of sync with the simulator even though there is no bug. The unpredictable reservation is designed to be temporary such that when a deterministic value is subsequently written to the register, FORCE unreserves the register.

Those are the major cases I can think of. There are a few other cases where FORCE may reserve a register for a limited period, such as when generating a loop. If the loop count register is modified within the loop, many issues can arise, e.g. a loop that never terminates.

Many tests that use the default comprehensive exception handlers would only observe 1 register reserved for the address table. The "fast" handler and address table registers are randomized such that different registers are reserved for each generated test. FORCE doesn't prevent the user from explicitly specifying a reserved register for use in an instruction. It does exclude them from being return by `getRandomRegister()` and `getRandomGPR()` and related methods in an effort to help the user avoid accidentally selecting one. If there are some specific scenarios that these mechanisms make impossible, we may be able to examine them and see if we can adjust the behavior to make them feasible to generate.

niuqiang-real, 2022-05-07:
Thanks for your detailed description about the reservation mechanism in FORCE. This really helps me a lot for understanding FORCE and using FORCE in my work!!!

noahsherrill, 2022-05-26:
You are welcome, @niuqiang-real. Please let me know if there are any other related issues or questions. Otherwise, I will close this issue soon.

niuqiang-real, 2022-05-30:
Thanks. I have no more question about this issue~

closed 2022-05-31 by noahsherrill

xusq123, 2024-02-29:
i want to know  where the code of exception handler

MikeOpenHWGroup, 2024-02-29:
Hi @xusq123.  This issue is closed and we try to ensure that we have one topic per GitHub Issue.   If you have a problem or question, please open a new Issue.  Also, try to provide as much information as you can to help us work with you to solve the problem.