openhwfoundation/force-riscv #58: Passing signed values in IMMediates not working

quswar-abid, 2022-05-31:
Following is a code for calculating factorial of a number:

```
addi x10 x0 0
addi x11 x0 1
addi x12 x0 2
addi x13 x0 3
addi x14 x0 4
addi x15 x0 5
addi x16 x0 6
addi x17 x0 7
addi x18 x0 5
addi x19 x0 25
addi x20 x0 20

auipc x5 0x10000
addi x5 x5 -44
lw x10 0 x5
jal x1 16
add x11 x0 x10
addi x10 x0 10
ecall
add x7 x10 x0
addi x10 x0 1
bne x7 x0 8
jal x0 20

beq x7 x0 16
mul x10 x10 x7
addi x7 x7 -1
jal x0 -12
jalr x0 x1 0
```

Which translates to:

```
        self.genInstruction("ADDI##RISCV", {"rd":10, "rs1":0, "simm12":0})#INT32(0)
        self.genInstruction("ADDI##RISCV", {"rd":11, "rs1":0, "simm12":1})
        self.genInstruction("ADDI##RISCV", {"rd":12, "rs1":0, "simm12":2})
        self.genInstruction("ADDI##RISCV", {"rd":13, "rs1":0, "simm12":3})
        self.genInstruction("ADDI##RISCV", {"rd":14, "rs1":0, "simm12":4})
        self.genInstruction("ADDI##RISCV", {"rd":15, "rs1":0, "simm12":5})
        self.genInstruction("ADDI##RISCV", {"rd":16, "rs1":0, "simm12":6})
        self.genInstruction("ADDI##RISCV", {"rd":17, "rs1":0, "simm12":7})
        self.genInstruction("ADDI##RISCV", {"rd":18, "rs1":0, "simm12":5})
        self.genInstruction("ADDI##RISCV", {"rd":19, "rs1":0, "simm12":25})
        self.genInstruction("ADDI##RISCV", {"rd":20, "rs1":0, "simm12":20})
        self.genInstruction("AUIPC##RISCV",{"rd":5, "simm20":0x10000})
        self.genInstruction("ADDI##RISCV", {"rd":5, "rs1":5, "simm12":-44})#"INT32(-44)"
        self.genInstruction("LW##RISCV",   {"rd":10, "rs1":0, "simm12":5})
        self.genInstruction("JAL##RISCV",  {"rd":1, "simm20":16})
        self.genInstruction("ADD##RISCV",  {"rd":11, "rs1":0, "rs2":10})
        self.genInstruction("ADDI#RISCV",  {"rd":10, "rs1":0, "rs2":10})
        self.genInstruction("ECALL##RISCV")
        self.genInstruction("ADD##RISCV",  {"rd":7, "rs1":10, "rs2":0})
        self.genInstruction("ADDI##RISCV", {"rd":10, "rs1":0, "simm12":1})
        self.genInstruction("BNE##RISCV",  {"rs1":7, "rs2":0, "simm12":8})
        self.genInstruction("JAL##RISCV",  {"rd":0, "simm20":20})
        self.genInstruction("BEQ##RISCV",  {"rs1":7, "rs2":0, "simm12":16})
        self.genInstruction("MUL##RISCV",  {"rd":10, "rs1":10, "rs2":7})
        self.genInstruction("ADDI##RISCV", {"rd":7, "rs1":7, "simm12":-1})
        self.genInstruction("JAL##RISCV",  {"rd":0, "simm20":-12})
        self.genInstruction("JALR##RISCV", {"rd":0, "rs1":1, "simm12":0})
        
```

The following CLI log shows the error in parsing signed Int. when compiled with following CLI command:

`bin/friscv -t utils/smoke/new_riscv_force.py -s 0x100 -c config/riscv_rv32.config`

Same has also been observed when using _rv64.config as well.
```

[notice]Generating: AUIPC##RISCV
[notice]{OperandGenerate} Operand : simm20 Constraint: 0x10000 value 0x65536
[notice]Committing instruction "AUIPC x5, 65536" at 0x8001102c=>[0]0x8001102c (0x10000297) gen(0)
[SimApiHANDCAR::WriteRegister] x5 0x49b635d6d73fd751/0xffffffffffffffff
[fail]Failed parsing 'INT32(', invalid argument.
[FAIL]{parse-error-uint64-invalid} in file '../base/src/StringUtils.cc' line 39 func 'parse_uint64'.
```

The examples provided in the document (FORCE-RISCV_User_Manual-v0.8.pdf) at page#20 (for genInstruction API) encounter the same issue.

quswar-abid, 2022-05-31:
Issue resolved, it was caused by a syntax error.

noahsherrill, 2022-05-31:
If my understanding is correct, the issue concerning the immediate operand override was due to a syntax error in the test template. It sounds as though you encountered another issue concerning the override for GPRs, but I am not sure what the issue is. Could you please clarify what the behavior you are seeing is and what the expected behavior is for the GPR override issue?

quswar-abid, 2022-06-01:
Thanks @noahsherrill for the response. Yes, initially it was a syntax error. 

What was supposed to be {"simm12.Data":"INT32(0)"} was written as {"simm12":"INT32(0)"} as I tried to just copy-paste and modify the examples from genInstruction front end API.

I expect this to set the immediate value to a fix "0" but this is not what's happening. It's a different value which I noticed is pseudo-random (it changes with the set seed value) and is totally random if no seed is provided. 

Kindly guide me on this one. Am I using the API wrong? What is the right way?

noahsherrill, 2022-06-03:
I'm not very familiar with the `.Data` syntax. As far as I can tell, it actually only has any effect for uninitialized register operands, where it can be used to specify an initial value for the register. The way to accomplish what you want is to use `{"simm12": 0}`.

The FORCE log file should indicate `ignored operand data override` whenever `.Data` is appended to an immediate operand name. There are also other instances in which an argument passed to `genInstruction()` is ignored, typically because FORCE is unable to satisfy the request. This can be easy to miss in the log output, so if you ever want to check whether any of the `genInstruction()` arguments are being ignored, you can pass `--failOverride` to FORCE. This makes FORCE fail whenever it is unable to generate with the specified arguments.

quswar-abid, 2022-06-06:
Thanks @noahsherrill for your response. Issue resolved!

closed 2022-06-06 by quswar-abid