openhwfoundation/force-riscv #60: Is there any plan to support hypervisor extension?

hanfeng0114, 2022-06-04:


noahsherrill, 2022-06-06:
We have discussed it and would like to add support for it. Unfortunately, we don't have the resources to work on it at present, so I don't know when or if we will be able implement it.

MikeOpenHWGroup, 2022-06-06:
Hi @hanfeng0114, thanks for your question.  As you may know, FORCE is an [OpenHW Group](https://www.openhwgroup.org/) project.  All our projects are supported by one or more members.  In the case of FORCE, Futurewei is the primary driver and to date has made all of the contributions.  Having said that, as an open source organization, we are open to contributions from all so if you or your company would be interested in helping to add hypervisor support to FORCE, this would be most welcome.

hanfeng0114, 2022-06-29:
Is it possible to provide some tips or feature list to support hypervisor features?
It is hard for new comers to list all related features.

Thanks

hanfeng0114, 2023-05-15:
Call for more people to get involved for this feature. Will discuss it later.

hanfeng0114, 2024-02-13:
Please join mattermost first.
Follow the steps to reigister eclipse account: https://www.openhwgroup.org/register/

Then join FORCE-RISCV VTG mattermost, which is openhw's official instant message communication tool: https://mattermost.openhwgroup.org/all-users/channels/vtg-force-riscv#

Stage2 development of FORCE-RISCV will be tracked in FORCE-RISCV VTG mattermost.

kong-ling-hui, 2024-04-11:
> We have discussed it and would like to add support for it. Unfortunately, we don't have the resources to work on it at present, so I don't know when or if we will be able implement it.

Hello, I would like to ask if you have evaluated the manpower and time required for development？