openhwfoundation/force-riscv #61: How to override the test termination method of FORCE-RISCV?

quswar-abid, 2022-06-08:
Force-RV marks the end of test with a "jump-to-self" instruction. How can we modify it so it suits our own testing criteria?

noahsherrill, 2022-06-08:
We've had some interest in being able to customize the end of test sequence, but no one has had the opportunity to implement such a feature as yet. If you are willing to modify the source code, you can change it in the `GenSequenceAgentRISCV::GetEndOfTestSequence()` method in `riscv/src/GenSequenceAgentRISCV.cc`. You can follow the other examples in that file to create the necessary instruction requests and push them into the `rReqSeq` vector.

MikeOpenHWGroup, 2022-06-09:
The ability to customize the end of test sequence is a high value feature.  Almost every testbench will implement its own strategy for this and being able to easy override a boiler-plate end-of-program is very important.

quswar-abid, 2022-06-09:
@noahsherrill thanks for responding. Yes, I took the liberty to go through that file, and I tried to modify the `BranchToSelf` routine called by `EndOfTestSequence`. My modification looks like this:

```
  void GenSequenceAgentRISCV::GetBranchToSelfSequence(vector<GenRequest* >& req_seq) const
  {
    // We avoid disabling BNT here to enable speculative reads past the end-of-test instruction
    //const char* j_instr = "JAL##RISCV";
    //auto j_req = new GenInstructionRequest(j_instr);
    //j_req->AddDetail("NoRestriction", 1);
    //j_req->AddOperandRequest("rd", 0);
    //j_req->AddOperandRequest("simm20", 0);
    //req_seq.push_back(j_req);
    
    //auipc	t0, 327680
    const char* j_instr_1 = "AUIPC##RISCV";
    auto j_req_1 = new GenInstructionRequest(j_instr_1);
    //j_req_1->AddDetail("NoRestriction", 1);
    j_req_1->AddOperandRequest("rd", "t0");
    j_req_1->AddOperandRequest("simm20", 327680);
    req_seq.push_back(j_req_1);
    
    //addi	t0, t0, -240
    const char* j_instr_2 = "ADDI##RISCV";
    auto j_req_1 = new GenInstructionRequest(j_instr_2);
    //j_req_2->AddDetail("NoRestriction", 1);
    j_req_2->AddOperandRequest("rd", "t0");
    j_req_2->AddOperandRequest("rs1", "t0");
    j_req_2->AddOperandRequest("simm12", -240);
    req_seq.push_back(j_req_2);
    
    //sd	a0, 0(t0)
    const char* j_instr_3 = "ADDI##RISCV";
    auto j_req_3 = new GenInstructionRequest(j_instr_3);
    //j_req_3->AddDetail("NoRestriction", 1);
    j_req_3->AddOperandRequest("rs1", "t0");
    j_req_3->AddOperandRequest("rs2", "a0");
    j_req_3->AddOperandRequest("simm12", 0);
    req_seq.push_back(j_req_3); 
       
    //j	0x800000e4 <_eoc>
    const char* j_instr_4 = "C.J##RISCV";
    auto j_req_4 = new GenInstructionRequest(j_instr_4);
    //j_req_3->AddDetail("NoRestriction", 1);
    j_req_4->AddOperandRequest("simm11", -12);
    req_seq.push_back(j_req_4);
  }
```

Unfortunately it did not had any effect on generated code. I am still seeing jump to self (`0000006f JAL x0, 0`) in the `*.Default.S `file. Any idea what additional steps I need to do?

_Note:_ 
Right now, I am trying to generate tests for CVA6 core through ARA repo. I saw that's how tests conclude according to the dump files in `ara/apps/bin` and so I tried to mimic that in generated tests or else the tests get stuck at jump to self instruction.
_One additional note:_
The configuration file for CVA6 looks stale. Maybe no one has tested it too, since further development in FORCE-RV project. We managed to put together a `.config` file but I'd be happy to have comments on it. I'll try to bring this in a separate issue later. Meanwhile, please provide comments on that as well.

MikeOpenHWGroup, 2022-06-09:
> The configuration file for CVA6 looks stale.

This is almost certainly the case. The CVA6 is not currently using FORCE for test-program generation.  Please create an issue for this and I will get you some support from the CVA6 team.

noahsherrill, 2022-06-09:
When I tested the above code, it didn't compile. I changed `auto j_req_1 = new GenInstructionRequest(j_instr_2);` to `auto j_req_2 = new GenInstructionRequest(j_instr_2);`, and then it compiles for me.

There are then a few issues with executing it. One is that FORCE doesn't understand the assembly register aliases, so I recommend changing `"t0"` to `5` and `"a0"` to `10`. It looks like `const char* j_instr_3 = "ADDI##RISCV";` is probably intended to be `const char* j_instr_3 = "SD##RISCV";`. The immediate operand values also need to be adjusted because the interface expresses them as C++ 64-bit unsigned integers, so `-240` needs to be changed to `0xF10` and `-12` to `0x7F4`. I recommend adding `j_req_3->AddDetail("NoPreamble", 1);` for the `SD` instruction; otherwise, FORCE may generate additional instructions to load a value into the base register.

I also noticed the comment for the last instruction is `//j	0x800000e4 <_eoc>` while the instruction used is `C.J`. Was it the intention to use `J` or `C.J`?

quswar-abid, 2022-06-10:
Yeah, I corrected the mistake earlier. I did made those changes and it did compile. However I have yet to try `NoPreamble` thing. Another thing I tried to end the code targeted for CVA6 was to add `ECALL`, but that also came along with around 60 extra instructions. What do we do about it?

quswar-abid, 2022-06-10:
@noahsherrill The `AddDetail("NoPreamble", 1);` did not prevent FORCE to insert additional instructions. Also, where can I find details regarding backend APIs? I generated the initially mentioned sequence from the jump-to-self as an example. Is there any document available?

quswar-abid, 2022-06-10:
@noahsherrill Since I did not found J in general instructions, I did it by using the instruction mentioned in compressed instructions. Should I add it in general instructions myself, or keep using compressed instruction? What do you suggest?

noahsherrill, 2022-06-10:
The details regarding the back-end APIs are in the source code; there is no additional documentation for them. The `unit_tests` directory contains some unit tests, which can serve as useful examples on how to use some of the methods.

I believe `J` is an assembly alias for `JAL` with `rd` set to `x0`. I probably need to see the exact code being executed, the test being run and the random seed being used to be able to reproduce the issue you are observing with `NoPreamble`. I'll take a look at `ECALL`.

noahsherrill, 2022-06-10:
I placed an `ECALL` in between the `SD` and the `JAL`, and I don't see any additional instructions being generated. There are cases where FORCE attempts to generate additional instructions for various reasons, but I can't think of any mechanism for this that would be relevant to the `ECALL` instruction. Are you seeing these additional instructions in the `Default.S` file immediately preceding the `ECALL` instruction?

quswar-abid, 2022-06-13:
As for the code, I am just creating 5 nopes, with seed (`0xea4507d444a3682f`) as in following command:
`
`$FORCE_PATH/bin/friscv -t $FORCE_PATH/empty_test.py --max-instr 50000 --num-chips 1 --num-cores 1 --num-threads 1 -s 0xea4507d444a3682f --cfg $FORCE_PATH/config/riscv-cva6.config`
`

`$FORCE_PATH` is the root folder containing force-riscv repo. The rest of changes in the backend are as following:

`
  void GenSequenceAgentRISCV::GetBranchToSelfSequence(vector<GenRequest* >& req_seq) const
  {
    // We avoid disabling BNT here to enable speculative reads past the end-of-test instruction
    /*
    const char* j_instr = "JAL##RISCV";
    auto j_req = new GenInstructionRequest(j_instr);
    j_req->AddDetail("NoRestriction", 1);
    j_req->AddOperandRequest("rd", 0);
    j_req->AddOperandRequest("simm20", 0);
    req_seq.push_back(j_req);
    

    const char* j_instr = "ECALL##RISCV";
    auto j_req = new GenInstructionRequest(j_instr);
    //j_req->AddDetail("NoRestriction", 1);
    //j_req->AddOperandRequest("rd", 0);
    //j_req->AddOperandRequest("simm20", 0);
    req_seq.push_back(j_req);
    */
    //auipc	t0, 327680
    const char* j_instr_1 = "AUIPC##RISCV";
    auto j_req_1 = new GenInstructionRequest(j_instr_1);
    j_req_1->AddDetail("NoRestriction", 1);
    j_req_1->AddOperandRequest("rd", 5);
    j_req_1->AddOperandRequest("simm20", 327680);
    req_seq.push_back(j_req_1);
    
    //addi	t0, t0, -240
    const char* j_instr_2 = "ADDI##RISCV";
    auto j_req_2 = new GenInstructionRequest(j_instr_2);
    j_req_2->AddDetail("NoRestriction", 1);
    j_req_2->AddOperandRequest("rd", 5);
    j_req_2->AddOperandRequest("rs1", 5);
    j_req_2->AddOperandRequest("simm12", 0xf10);//-240
    req_seq.push_back(j_req_2);
    
    //sd	a0, 0(t0)
    const char* j_instr_3 = "SD##RISCV";
    auto j_req_3 = new GenInstructionRequest(j_instr_3);
    j_req_3->AddDetail("NoRestriction", 1);
    j_req_3->AddDetail("NoPreamble", 1);
    j_req_3->AddOperandRequest("rs1", 5);
    j_req_3->AddOperandRequest("rs2", 10);
    j_req_3->AddOperandRequest("simm12", 0);
    req_seq.push_back(j_req_3); 
       
    //j	0x800000e4 <_eoc>
    const char* j_instr_4 = "JAL##RISCV";
    auto j_req_4 = new GenInstructionRequest(j_instr_4);
    j_req_4->AddDetail("NoRestriction", 1);
    j_req_4->AddOperandRequest("rd", 0);
    j_req_4->AddOperandRequest("simm20", 0xffffc);
    req_seq.push_back(j_req_4);
    
  }

`

noahsherrill, 2022-06-16:
I have run the given code with the given seed and with some other seeds, and I do not see any instructions being generated between `ADDI` and `SD`. When I uncomment the `ECALL`, I also don't see any instructions generated between the last `NOP` of the test and `ECALL`.

I see the following in the `Default.S` file starting from `0x80011000`:
Without `ECALL`:
```
0000000080011000:00000013 ADDI x0, x0, 0
0000000080011004:00000013 ADDI x0, x0, 0
0000000080011008:00000013 ADDI x0, x0, 0
000000008001100c:00000013 ADDI x0, x0, 0
0000000080011010:00000013 ADDI x0, x0, 0
0000000080011014:50000297 AUIPC x5, 327680
0000000080011018:f1028293 ADDI x5, x5, -240
000000008001101c:00a2b023 SD x10, x5, 0
0000000080011020:ff9ff06f JAL x0, -4
0000000080011024:00710eb3 ADD x29, x2, x7
```
With `ECALL`:
```
0000000080011000:00000013 ADDI x0, x0, 0
0000000080011004:00000013 ADDI x0, x0, 0
0000000080011008:00000013 ADDI x0, x0, 0
000000008001100c:00000013 ADDI x0, x0, 0
0000000080011010:00000013 ADDI x0, x0, 0
0000000080011014:00000073 ECALL
0000000080011018:50000297 AUIPC x5, 327680
000000008001101c:f1028293 ADDI x5, x5, -240
0000000080011020:00a2b023 SD x10, x5, 0
0000000080011024:ff9ff06f JAL x0, -4
0000000080011028:00710eb3 ADD x29, x2, x7
```
I added a couple more lines of code before the `SD` instruction:
```
uint64 target_addr = mpGenerator->PC() + (327680 << 12) - 240;
mpGenerator->InitializeMemory(target_addr, 0, 8, 0, false, true);
```
Because we are generating `SD` with `NoRestriction`, which skips a lot of processing, we need to make sure the memory is initialized at the target address.

As I examine this sequence more, it seems strange to me. The target address for the `SD` instruction is variable because it's based on the PC at the end of the test, which could be any legal address depending on the instructions used in the test. Why would a write to a random address be used as part of the end of test sequence? Also, why is the final instruction jumping two instructions back?

noahsherrill, 2022-06-16:
Note that the value returned by `mpGenerator->PC()` is the value of the PC at the beginning of the sequence, so if any other instructions, such as `ECALL`, are added prior to `AUIPC`, the target address calculation needs to be adjusted accordingly.

quswar-abid, 2022-08-11:
In the `riscv/src/GenSequenceAgentRISCV.cc`, I have updated the `GetEndOfTestSequence` sequence as following:

```
  void GenSequenceAgentRISCV::GetEndOfTestSequence(vector<GenRequest*>& rReqSeq) const
  {
    //GetBranchToSelfSequence(rReqSeq);
    GetBranchToSelfSequence_modified(rReqSeq);
  }
```

Thereby calling;
```
  void GenSequenceAgentRISCV::GetBranchToSelfSequence_modified(vector<GenRequest* >& req_seq) const
  {
    // We avoid disabling BNT here to enable speculative reads past the end-of-test instruction
    
    //ADDI t0, t0, 0x5
    const char* j_instr_2 = "ADDI##RISCV";
    auto j_req_2 = new GenInstructionRequest(j_instr_2);
    j_req_2->AddDetail("NoRestriction", 1);
    j_req_2->AddOperandRequest("rd", 5);
    j_req_2->AddOperandRequest("rs1", 0);
    j_req_2->AddOperandRequest("simm12", 0xd);
    req_seq.push_back(j_req_2);
    
    //SLLI t0, t0, 28
    const char* j_instr_3 = "SLLI#RV64I#RISCV";
    auto j_req_3 = new GenInstructionRequest(j_instr_3);
    j_req_3->AddDetail("NoRestriction", 1);
    j_req_3->AddOperandRequest("rd", 5);
    j_req_3->AddOperandRequest("rs1", 5);
    j_req_3->AddOperandRequest("shamt", 28);
    req_seq.push_back(j_req_3);
    
    //ADDI a10, x0, 0
    const char* j_instr_4 = "ADDI##RISCV";
    auto j_req_4 = new GenInstructionRequest(j_instr_4);
    j_req_4->AddDetail("NoRestriction", 1);
    j_req_4->AddOperandRequest("rd", 10);
    j_req_4->AddOperandRequest("rs1", 0);
    j_req_4->AddOperandRequest("simm12", 0);
    req_seq.push_back(j_req_4);
    
    //SD a0, 0(t0)
    const char* j_instr_5 = "SD##RISCV";
    auto j_req_5 = new GenInstructionRequest(j_instr_5);
    j_req_5->AddDetail("NoRestriction", 1);
    j_req_5->AddDetail("NoPreamble", 1);
    j_req_5->AddDetail("NoSkip", 1);
    j_req_5->AddOperandRequest("rs1", 5);
    j_req_5->AddOperandRequest("rs2", 10);
    j_req_5->AddOperandRequest("simm12", 0);
    req_seq.push_back(j_req_5);
    
    //JAL 0, -4
    const char* j_instr_6 = "JAL##RISCV";
    auto j_req_6 = new GenInstructionRequest(j_instr_6);
    j_req_6->AddDetail("NoRestriction", 1);
    j_req_6->AddOperandRequest("rd", 0);
    //j_req_6->AddOperandRequest("rs1", 0);
    j_req_6->AddOperandRequest("simm20", 0xffffe);//0xffffc=-4
    req_seq.push_back(j_req_6);
    /**/
  }
```
I also added it (the prototype) in the `riscv/inc/GenSequenceAgentRISCV.h` file.

Now if I have to store `0` to address `0xd0000000` I need to initiliaze it first. I do have an address generation API in front end as following:

```
class MainSequence(Sequence):
    def generate(self, **kargs):
         self.initializeMemory(0xd0000000, 0, 8, 0, False, False)
```

But I need an API in the back end to include the inialization of adress `0xd0000000` in the backend. How can I do that?

quswar-abid, 2022-08-11:
Thanks @noahsherrill, would you be kind enough to guide me on this one:

In the `riscv/src/GenSequenceAgentRISCV.cc`, I have updated the `GetEndOfTestSequence` sequence as following:

```
  void GenSequenceAgentRISCV::GetEndOfTestSequence(vector<GenRequest*>& rReqSeq) const
  {
    //GetBranchToSelfSequence(rReqSeq);
    GetBranchToSelfSequence_modified(rReqSeq);
  }
```

Thereby calling;
```
  void GenSequenceAgentRISCV::GetBranchToSelfSequence_modified(vector<GenRequest* >& req_seq) const
  {
    // We avoid disabling BNT here to enable speculative reads past the end-of-test instruction
    
    //ADDI t0, t0, 0x5
    const char* j_instr_2 = "ADDI##RISCV";
    auto j_req_2 = new GenInstructionRequest(j_instr_2);
    j_req_2->AddDetail("NoRestriction", 1);
    j_req_2->AddOperandRequest("rd", 5);
    j_req_2->AddOperandRequest("rs1", 0);
    j_req_2->AddOperandRequest("simm12", 0xd);
    req_seq.push_back(j_req_2);
    
    //SLLI t0, t0, 28
    const char* j_instr_3 = "SLLI#RV64I#RISCV";
    auto j_req_3 = new GenInstructionRequest(j_instr_3);
    j_req_3->AddDetail("NoRestriction", 1);
    j_req_3->AddOperandRequest("rd", 5);
    j_req_3->AddOperandRequest("rs1", 5);
    j_req_3->AddOperandRequest("shamt", 28);
    req_seq.push_back(j_req_3);
    
    //ADDI a10, x0, 0
    const char* j_instr_4 = "ADDI##RISCV";
    auto j_req_4 = new GenInstructionRequest(j_instr_4);
    j_req_4->AddDetail("NoRestriction", 1);
    j_req_4->AddOperandRequest("rd", 10);
    j_req_4->AddOperandRequest("rs1", 0);
    j_req_4->AddOperandRequest("simm12", 0);
    req_seq.push_back(j_req_4);
    
    //SD a0, 0(t0)
    const char* j_instr_5 = "SD##RISCV";
    auto j_req_5 = new GenInstructionRequest(j_instr_5);
    j_req_5->AddDetail("NoRestriction", 1);
    j_req_5->AddDetail("NoPreamble", 1);
    j_req_5->AddDetail("NoSkip", 1);
    j_req_5->AddOperandRequest("rs1", 5);
    j_req_5->AddOperandRequest("rs2", 10);
    j_req_5->AddOperandRequest("simm12", 0);
    req_seq.push_back(j_req_5);
    
    //JAL 0, -4
    const char* j_instr_6 = "JAL##RISCV";
    auto j_req_6 = new GenInstructionRequest(j_instr_6);
    j_req_6->AddDetail("NoRestriction", 1);
    j_req_6->AddOperandRequest("rd", 0);
    //j_req_6->AddOperandRequest("rs1", 0);
    j_req_6->AddOperandRequest("simm20", 0xffffe);//0xffffc=-4
    req_seq.push_back(j_req_6);
    /**/
  }
```
I also added it (the prototype) in the `riscv/inc/GenSequenceAgentRISCV.h` file.

Now if I have to store `0` to address `0xd0000000` I need to initiliaze it first. I do have an address generation API in front end as following:

```
class MainSequence(Sequence):
    def generate(self, **kargs):
         self.initializeMemory(0xd0000000, 0, 8, 0, False, False)
```

But I need an API in the back end to include the inialization of adress `0xd0000000` in the backend. How can I do that?

Also @noahsherrill how can I easily obtain more information on backend APIs. Is there a document that can identify the mapping between backend and frontend APIs? I'd really appreciate your response.

quswar-abid, 2022-08-12:
Thank you for responding, the issue is resolved on my side, I'll close the issue now.

closed 2022-08-12 by quswar-abid