openhwfoundation/force-riscv #62: The configuration file for CVA6 needs update

quswar-abid, 2022-06-10:
@MikeOpenHWGroup I see the configuration file for CVA6 is there, but might not have been tested with further development in the repository. I have a version of file that is updated to work with CVA6. Right now I am generating code with it to run on ARA. [I put it on a gist](https://gist.github.com/quswar-abid/555f7fc7a36da92b4fe2fb320c69e4b4) as of now, but can push through a PR if I encounter no further errors. Another thing that needs update is the update related to addressable memory. One in `py/riscv/memory-cva6.py` and one in `py/riscv/PcConfig.py` which are as following:

In memory-cva6, following function is modified:
```
def configure_memory(interface):
    interface.addMemoryRange(0, 0x000080000000, 0x000080100000)
```

And the PcConfig has following updates to satisfy the assertions in ARA:
```
def get_reset_pc():
    return 0x80000000

def get_initial_pc(thread_id):
    return get_base_initial_pc() + thread_id * get_initial_pc_offset()

def get_boot_pc(thread_id):
    return get_base_boot_pc() + thread_id * get_boot_pc_offset()

def get_reset_region_size():
    return 0x40

def get_boot_region_size():
    return 0x1000

def get_base_initial_pc():
    return 0x80011000

def get_base_boot_pc():
    return 0x80000024

def get_initial_pc_offset():
    return 0x1000000

def get_boot_pc_offset():
    return 0x100000
```

@noahsherrill can you test the given changes?

quswar-abid, 2022-06-14:
[PR created for CVA6](https://github.com/openhwgroup/force-riscv/pull/63).

quswar-abid, 2022-06-14:
#63

quswar-abid, 2022-08-12:
Made the required updates in Pull Request #63 
Kindly review.

hanfeng0114, 2023-05-06:
Hi @MikeOpenHWGroup, it seems the user account is deleted, and it there any new plan for cva6 support? If not, can we close this issue?

MikeOpenHWGroup, 2023-05-06:
Agreed.  We can close this issue.

closed 2023-05-06 by MikeOpenHWGroup