openhwfoundation/force-riscv #64: Generate a .elf 32 bits is impossible?

EdgeOfMemory-cloud, 2022-06-23:
Good morning,

Since the last week I have downloaded and used your generator which is very useful but I have a problem. My personal ISS needs a 32 bits .elf file to work. I use the command **./bin/friscv -t <test_template> --cfg config/riscv_rv32.config** with my test template but the .elf file in the end is still 64 bits and useless. Is it a bug or do I missed something?
For example the command **fileriscv_force.Default.ELF** gives **riscv_force.Default.ELF: ELF 64-bit LSB executable, UCB RISC-V, version 1 (GNU/Linux), statically linked, stripped**.

Thank you in advance for your answer.

Louis-Gabriel

rchu-futurewei, 2022-06-25:
FORCE-RISCV uses ELFIO to write out the elf file.  However, it is currently fixed to ELF64.  But it shouldn't be too difficult  to add in options and make the necessary modifications to write out to ELF32 format, assuming addressing are limited to 32-bits.  Unfortunately, our resources are limited to make this happen.  I would encourage the community to be brave to provide code to extend the features of FORCE.

In this case, in the file base/src/TestIO.cc is where one can start, specifically the DumpToElf() function, to make the changes to support also ELF32.

quswar-abid, 2022-08-01:
> I would encourage the community to be brave to provide code to extend the features of FORCE.

The frontend APIs are some what very well documented. Documenting back-end enough would also be a help to enable that ☝️☝️☝️

hanfeng0114, 2023-05-15:
@EdgeOfMemory-cloud is there any question for this issue?

closed 2023-05-15 by EdgeOfMemory-cloud

xusq123, 2024-09-24:
> FORCE-RISCV uses ELFIO to write out the elf file. However, it is currently fixed to ELF64. But it shouldn't be too difficult to add in options and make the necessary modifications to write out to [...]
how to add write_to_host and from_to_host which function is to edit