openhwfoundation/force-riscv #66: Backend APIs for Memory Initialization

quswar-abid, 2022-08-12:
Hi @noahsherrill, would you be kind enough to guide me on this one:

In the `riscv/src/GenSequenceAgentRISCV.cc`, I have updated the `GetEndOfTestSequence` sequence as following:

```
  void GenSequenceAgentRISCV::GetEndOfTestSequence(vector<GenRequest*>& rReqSeq) const
  {
    //GetBranchToSelfSequence(rReqSeq);
    GetBranchToSelfSequence_modified(rReqSeq);
  }
```

Thereby calling;
```
  void GenSequenceAgentRISCV::GetBranchToSelfSequence_modified(vector<GenRequest* >& req_seq) const
  {
    // We avoid disabling BNT here to enable speculative reads past the end-of-test instruction
    
    //ADDI t0, t0, 0x5
    const char* j_instr_2 = "ADDI##RISCV";
    auto j_req_2 = new GenInstructionRequest(j_instr_2);
    j_req_2->AddDetail("NoRestriction", 1);
    j_req_2->AddOperandRequest("rd", 5);
    j_req_2->AddOperandRequest("rs1", 0);
    j_req_2->AddOperandRequest("simm12", 0xd);
    req_seq.push_back(j_req_2);
    
    //SLLI t0, t0, 28
    const char* j_instr_3 = "SLLI#RV64I#RISCV";
    auto j_req_3 = new GenInstructionRequest(j_instr_3);
    j_req_3->AddDetail("NoRestriction", 1);
    j_req_3->AddOperandRequest("rd", 5);
    j_req_3->AddOperandRequest("rs1", 5);
    j_req_3->AddOperandRequest("shamt", 28);
    req_seq.push_back(j_req_3);
    
    //ADDI a10, x0, 0
    const char* j_instr_4 = "ADDI##RISCV";
    auto j_req_4 = new GenInstructionRequest(j_instr_4);
    j_req_4->AddDetail("NoRestriction", 1);
    j_req_4->AddOperandRequest("rd", 10);
    j_req_4->AddOperandRequest("rs1", 0);
    j_req_4->AddOperandRequest("simm12", 0);
    req_seq.push_back(j_req_4);
    
    //SD a0, 0(t0)
    const char* j_instr_5 = "SD##RISCV";
    auto j_req_5 = new GenInstructionRequest(j_instr_5);
    j_req_5->AddDetail("NoRestriction", 1);
    j_req_5->AddDetail("NoPreamble", 1);
    j_req_5->AddDetail("NoSkip", 1);
    j_req_5->AddOperandRequest("rs1", 5);
    j_req_5->AddOperandRequest("rs2", 10);
    j_req_5->AddOperandRequest("simm12", 0);
    req_seq.push_back(j_req_5);
    
    //JAL 0, -4
    const char* j_instr_6 = "JAL##RISCV";
    auto j_req_6 = new GenInstructionRequest(j_instr_6);
    j_req_6->AddDetail("NoRestriction", 1);
    j_req_6->AddOperandRequest("rd", 0);
    //j_req_6->AddOperandRequest("rs1", 0);
    j_req_6->AddOperandRequest("simm20", 0xffffe);//0xffffc=-4
    req_seq.push_back(j_req_6);
    /**/
  }
```
I also added it (the prototype) in the `riscv/inc/GenSequenceAgentRISCV.h` file.

Now if I have to store `0` to address `0xd0000000` I need to initiliaze it first. I do have an address generation API in front end as following:

```
class MainSequence(Sequence):
    def generate(self, **kargs):
         self.initializeMemory(0xd0000000, 0, 8, 0, False, False)
```

But I need an API in the back end to include the inialization of adress `0xd0000000` in the backend. How can I do that?

Also @noahsherrill how can I easily obtain more information on backend APIs. Is there a document that can identify the mapping between backend and frontend APIs? I'd really appreciate your response.

hanfeng0114, 2023-05-15:
It seems u want to use sd to a tube address to stop the simulation.
One workaround is that use genVAforPA to get one VA for 0xd0000000, then sd with that VA.

What's more, arch_data/general/general_choices.xml has one choice for your requirement, which is not well supported in the current release right now.
[image]

Will check whether to implement it at stage 2 work.

hanfeng0114, 2023-05-15:
@MikeOpenHWGroup it seems the user acount has been deleted. Please close this issue.

MikeOpenHWGroup, 2023-05-16:
Thanks for the update @hanfeng0114.  Closing.

closed 2023-05-16 by MikeOpenHWGroup

xusq123, 2024-02-01:
can not work
[image]