openhwfoundation/force-riscv #68: AccessFaultModifier not implemented

ProssMou, 2022-12-16:
I noticed inside the paging test folder, the access_fault_on_branch.py needs the AccessFaultModifier.
Hoever, it seems that the AccessFaultModifer has not implemented yet.
Does it mean the user should implemented it by themselves to verfiy access fault related feature?

hanfeng0114, 2023-05-06:
PMP is not currently well supported in current release.
But there is one workaround method. Take data pmp access fault as an example:
1. Initially, pmpcfg register fields should all be set as legal
2. in your template, use genPA to allocate one memory address range(say 4K), and then use genVAforPA to create pages
3. change pmp address register value and pmpcfg value for the allocated memory address range in step2

hanfeng0114, 2023-05-15:
@MikeOpenHWGroup It is the issue from Alibaba Xuantie team. U can close it right now. Will give some demo test on stage2 work.

MikeOpenHWGroup, 2023-05-16:
Thanks for the update @hanfeng0114.  Closing.

closed 2023-05-16 by MikeOpenHWGroup