openhwfoundation/force-riscv #70: memory of force-riscv generated elf files

Imperas, 2023-02-15:
We are attempting to target a real platform which has memory available at certain address ranges.
We have observed that force-riscv attempts to put memory regions in the full address range of the RV64GC processor...

How do we configure force-riscv to target our available memory ranges.
We cannot seem to find the options to control the memory address map...

[Simon]

rchu-futurewei, 2023-02-16:
The memory range can be specified with the "memory_file" option in the config file.  There is an example for cva6:
config/riscv-cva6.config
py/riscv/memory-cva6.py

Imperas, 2023-02-17:
Yes thanx - but this still creates a 1.5Gbyte memory binary file that is not easy to read into a Verilog test bench.
for example, force-riscv seems to generate some code at 50000000 and some at 80000000
how do I lose the boot code at 50000000?

and is there a way to get all the code that is generated to be contiguous from @80000000

thnx
[Simon]

rchu-futurewei, 2023-02-20:
To not generate boot code the commandline option "SkipBootCode=1" can be used.

There are a few parameters to control the reset, boot, and initial pc in the py/riscv/PcConfig.py file.  Reset PC is currently at 0x5000_0000, Boot code PC at 0x8000_0000, and Initial PC (where the test starts) at 0x8001_1000.

However, modifying PcConfig.py would only be a workaround.  A better solution would be to setup a few platform specific py layer files from which the test template should import from.

Specifically, for example, in the py/riscv directory copy:
EnvRISCV.py to EnvRISCVFoo.py
GenThreadRISCV.py to GenThreadRISCVFoo.py
ThreadSplitterSequence.py to ThreadSplitterSequenceFoo.py
PcConfig.py to PcConfigFoo.py

Then update PcConfigFoo.py accordingly for the particular platform for 
 the reset/boot/initial PCs.  And in each of the new *Foo.py files change the "import riscv.PcConfig" to "import riscv.PcConfigFoo".  Also, in EnvRISCVFoo to import riscv.ThreadSplitterSequenceFoo.py.

Finally, in your test template, change the imports of the EnvRISCV to be from riscv.EnvRISCVFoo, and GenThreadRISCV to be from riscv.EnvRISCVFoo.

hanfeng0114, 2023-05-15:
@Imperas can we close this issue?