openhwfoundation/force-riscv #72: handcar_cosim.so is too big

hanfeng0114, 2023-05-30:
Currently handcar_cosim.so is pushed into the force-riscv repo directly. The binary file is too big.
Start this issue to discuss possible maintenance policy like submodule?
Call for more discussions.

MikeOpenHWGroup, 2023-05-30:
Yes, 97M is a little big.  :smile:

Perhaps we can create a Makefile rule to allow users to compile it from source?  We do that for the `dpi_dasm` shared object in CORE-V-VERIF. (Although because that so is only 9.7M we also include it in the repo.)

hanfeng0114, 2023-06-04:
Start with some possible solutions:
1.google-repo (one multi-repo management solution which is used for android project maintenance), which is T-HEAD XUANTIE's inhouse solution
2.git submodule: an easy solution to attach one repo in another repo
3.mix with git submodule and google-repo: we may include other repo like riscv-opcode which helps spike development

hanfeng0114, 2023-06-05:
Will start a meeting recently about this issue. @MikeOpenHWGroup @DBees @rchu-futurewei