openhwfoundation/force-riscv #76: Compilation and simulation issue

kobeyu, 2023-06-20:
Hi,
I have two questions.

1.I'm using gcc-11 and Python 3.8 to compile this project, but I encountered the following error message (*1). It seems to be caused by the newer version of gcc because I can compile it successfully with gcc 9.3. I'm wondering if you have any plans to support newer versions of gcc.

2.I'm running the test_force.Default.ELF generated by gcc 9.3 on Spike, but I'm getting a "memory address is invalid" error. I would like to know if there is any way to fix this issue.

Access exception occurred while loading payload ./test_force.Default.ELF:
Memory address 0x50000000 is invalid

(*1)
../base/src/GenInstructionAgent.cc:394:21: error: loop variable ‘update’ creates a copy from type ‘const Force::RegUpdate’ [-Werror=range-loop-construct]
  394 |     for (auto const update : regUpdates) {
      |                     ^~~~~~
../base/src/GenInstructionAgent.cc:394:21: note: use reference type to prevent copying
  394 |     for (auto const update : regUpdates) {
      |                     ^~~~~~
      |                     &
g++ -c -Wall -std=c++11 -gdwarf-3 -m64 -Werror -O2 -Weffc++ -fvisibility=hidden -D 'ARCH_ENUM_HEADER="EnumsRISCV.h"' -I./inc -I../base/inc -I../3rd_party/inc -I/usr/include/python3.8 -o make_area/obj/ImageIO.o ../base/src/ImageIO.cc
In file included from ../base/src/ImageIO.cc:16:
../base/inc/ImageIO.h:39:38: error: ‘string’ in namespace ‘std’ does not name a type
   39 |     void PrintMemoryImage(const std::string& imageFile, const Memory* memory);  //!< write memory initial data to an image file.
      |                                      ^~~~~~
../base/inc/ImageIO.h:22:1: note: ‘std::string’ is defined in header ‘<string>’; did you forget to ‘#include <string>’?
   21 | #include "Defines.h"
  +++ |+#include <string>
   22 | 
../base/src/ImageIO.cc:517:8: error: no declaration matches ‘void Force::ImageIO::PrintRegistersImage(const string&, const std::map<std::__cxx11::basic_string<char>, long long unsigned int>&, const Force::RegisterFile*)’
  517 |   void ImageIO::PrintRegistersImage(const string& imageFile, const map<string, uint64>& threadInfo, const RegisterFile* regFile)
      |        ^~~~~~~
In file included from ../base/src/ImageIO.cc:16:
../base/inc/ImageIO.h:40:10: note: candidate is: ‘void Force::ImageIO::PrintRegistersImage(const int&, const int&, const Force::RegisterFile*)’
   40 |     void PrintRegistersImage(const std::string& imageFile, const std::map<std::string, uint64>& threadInfo, const RegisterFile* regFile); //!< write registers initial value to an Text file.
      |          ^~~~~~~~~~~~~~~~~~~
In file included from ../base/src/ImageIO.cc:16:
../base/inc/ImageIO.h:33:9: note: ‘class Force::ImageIO’ defined here
   33 |   class ImageIO {
      |

hanfeng0114, 2023-06-20:
For question 1:
There is no contest info like what os, which gcc version .etc.
But cmake build system is ready now. U can try cmake system to make sure whether it is gcc version problem.
''''
source setupenv.bash
mkdir build; cd build
make install -j

If there is problem again, replay the issue.

For question 2:
There is also no context info like which os, which version of spike. So I can't give proper suggestion.
But note that force-riscv generated ELF files has an assumption about memory map, which means maybe official spike model can't run the ELF file, because 0x50000000(Start address of FORCE-RISCV test) is a vacant location.

kobeyu, 2023-06-20:
Hi @hanfeng0114 

Thanks for your reply. The OS, GCC, and Spike information in (*1).

question 1: It's still not working, despite following these steps:
source setenv.bash                                                                                                             
mkdir build; cd build
cmake ..
make -j8

question 2: I'm wondering if you have any suggestions to fix this issue.

(*1)
os:
Linux geralt 5.15.0-69-generic #76-Ubuntu SMP Fri Mar 17 17:19:29 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

gcc:
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

spike:
commit c636ad356c3d5fd7d5ee565c59ab7bdc3f3852f5 (HEAD -> master, origin/master, origin/HEAD)
Merge: 2f994c5f cfe79e06
Author: Jerry Zhao <jerryz123@berkeley.edu>
Date:   Mon Jun 12 23:40:39 2023 -0700

    Merge pull request #1377 from riscv-software-src/ci-thorough
    
    Have CI run on each commit in a PR, instead of just the HEAD

hanfeng0114, 2023-06-21:
Question1: please show me the cmake build log info

Question2: the generated elf is not expected to run on the spike which is generated from official repo. force-riscv has resoncstruct spike as fpix_riscv, which is the target to execute the generated elf files

kobeyu, 2023-06-22:
Hi @hanfeng0114 

Question 1:Please refer to the link: https://drive.google.com/file/d/1k7nrGrnxk26KXsRuKpoH_ogVWxf9OU7A/view?usp=sharing
Question 2:Thanks, I get it. Actually, we plan to run .ELF on Spike and Gem5. The main purpose is to use it as a regression test after modifying the simulator. Have you ever tried running a RISC-V .ELF on Gem5?

hanfeng0114, 2023-06-26:
Question1: It seems the former coding style has some problem. Please have a try to add the folling line to "base/inc/ImageIO.h":
[image]
Your feedback is expected for the further enhancement.

Question2: Actually, the main problem is only memory map. The default memory map of FORCE-RISCV is:
0x50000000-0xFFFFFFFFFFFF.
U can change PcConfig.py and memory.py locally to change to expected memory map.
Please refer to #42 for further info.

hanfeng0114, 2023-06-26:
#70

kobeyu, 2023-06-29:
Hi @hanfeng0114 

Question 1: There is another issue when I added include string

[image]」

Question 2: I'll try it. thanks.

hanfeng0114, 2023-06-30:
Follow the steps to reigister eclipse account and join openhw mattermost:
https://www.openhwgroup.org/register/
https://mattermost.openhwgroup.org/all-users/channels/vtg-force-riscv#

Please join mattermost, we can chat there more timely.

hanfeng0114, 2023-07-17:
Hi, can u show me the cmake log, but not the make log?