openhwfoundation/force-riscv #77: Build Failed at Centos 7.9.2009

superbchetan, 2023-07-06:
On CentOS Linux release 7.9.2009 (Core). Cloned the repo (HEAD is at https://github.com/openhwgroup/force-riscv/commit/144fb52a99cde89e73552f88c872c05d2e90b603).

```
export FORCE_CC=/usr/bin/g++
export FORCE_PYTHON_VER=3.6
export FORCE_PYTHON_LIB=/apps/gcc/8.2.0/libexec/gcc/x86_64-pc-linux-gnu
export FORCE_PYTHON_INC=/usr/include/python3.6m
make
```

Getting following error

```
/usr/bin/g++ -c -Wall -std=c++11 -gdwarf-3 -m64 -Werror -O2 -Weffc++ -fvisibility=hidden -D 'ARCH_ENUM_HEADER="EnumsRISCV.h"' -I./inc -I../base/inc -I../3rd_party/inc -I/usr/include/python3.6m -o make_area/obj/TopLevelResources.o ../base/src/TopLevelResources.cc
../base/src/TopLevelResources.cc:50:10: error: base class ‘struct option::Arg’ has a non-virtual destructor [-Werror=effc++]
   struct Arg: public option::Arg
          ^
cc1plus: all warnings being treated as errors
make[3]: *** [make_area/obj/TopLevelResources.o] Error 1

```

Can someone please help to fix this error?

hanfeng0114, 2023-07-10:
Follow the steps to reigister eclipse account and join openhw mattermost:
https://www.openhwgroup.org/register/
https://mattermost.openhwgroup.org/all-users/channels/vtg-force-riscv#

Please join mattermost, we can chat there more timely.

hanfeng0114, 2023-07-10:
We tried centos linux release 7.8.2003 and gcc 7.4.0.
There is no resources to restore your scenario.
Hope we can discuss it in mattermost.

hanfeng0114, 2023-07-12:
Please review the suggestions of issue #76 .
If not fixed, please replay this issue with the cmake build error message.
Then we can make a meeting about it by mattermost.