openhwfoundation/force-riscv #80: Need help configuring to only generate RV32I

saw235, 2023-07-27:
Hi, I need some help to understand how to generate RV32I instructions

hanfeng0114, 2023-07-31:
I guess u need not only RV32I only, but also M mode only. Please refer to #37 first.
What's more, there is no resource to help to change the configurations, u can try yourself:
1. change config/riscv_rv32.config, change simulator_config_string as expected
2. change fpic/config/riscv_rv32.config, change simulator_config_string as expected
3. change as #37 details

If it works, u need to write your RV32I only templates.

Actually force-riscv is excepted to be used for heavy scale cpu verification, which need more infra and template development work. For some requirements, u need to change by yourself.

hanfeng0114, 2023-08-30:
Any more questions @saw235 ?

saw235, 2023-10-02:
A few followup questions, the instruction generator generated more than 500 instructions but in the sim.log, it is only running about 75 instructions which is unclear to me why that is happening.

hanfeng0114, 2023-10-03:
Template gerenation flow and execution flow are different flows. Sim.log is generated on generation flow, which will do some thread partition and reloading process. Check fpix_sim.log for the execution flow.

saw235, 2023-10-05:
I don't see any fpix_sim.log

saw235, 2023-10-05:
A few questions:

1. When i disassemble the generated elf file, i am seeing that the address is going to some random range. Is there a way to control where it starts and where it gets generated?

Singularity> riscv32-unknown-elf-objdump -d bazel-bin/riscv-rand-gen/gen_rand_test2-rand_gen.elf 

2. I also see that when i disassemble it with riscv32-unknown-elf-objdump 
I am seeing the format as 
bazel-bin/riscv-rand-gen/gen_rand_test2-rand_gen.elf:     file format elf64-littleriscv

Is there a way to get elf32?

saw235, 2023-10-05:
Im also seeing the following when dissasembling the elf files

hanfeng0114, 2023-10-07:
Please refer to #64 first.
Currently force-riscv ImanageIO just configure 64-bit version.

closed 2023-10-21 by hanfeng0114