openhwfoundation/force-riscv #81: Compilation Errors Occurred Building and Debugging the Project in VSCode

gcc-hub, 2023-08-26:
this is setting.json:
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceFolder}/**",
                "D:\\VScode\\Python3.6\\include"
            ],
            "defines": [
                "_DEBUG",
                "UNICODE",
                "_UNICODE"
            ],
            "compilerPath": "D:\\VScode\\mingw64\\bin\\gcc.exe",
            "cStandard": "c11",
            "cppStandard": "c++11",
            "intelliSenseMode": "windows-gcc-x64"
        }
    ],
    "version": 4
}

PS C:\Users\gageg\Desktop\force-riscv> cd .\build\
PS C:\Users\gageg\Desktop\force-riscv\build> cmake ..
CMake Deprecation Warning at CMakeLists.txt:15 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at 3rd_party/CMakeLists.txt:8 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at 3rd_party/tools/pybind11Tools.cmake:8 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  3rd_party/CMakeLists.txt:33 (include)

CMake Warning (dev) at 3rd_party/tools/FindPythonLibsNew.cmake:60 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  3rd_party/tools/pybind11Tools.cmake:16 (find_package)
  3rd_party/CMakeLists.txt:33 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- pybind11 v2.4.3
CMake Deprecation Warning at riscv/CMakeLists.txt:15 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at utils/handcar/CMakeLists.txt:15 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at fpix/CMakeLists.txt:15 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at fpix/riscv/CMakeLists.txt:15 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at unit_tests/CMakeLists.txt:15 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at unit_tests/tests/base/Memory/CMakeLists.txt:15 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at unit_tests/tests/fpix/EnumsFPIX/CMakeLists.txt:15 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at unit_tests/tests/riscv/EnumsRISCV/CMakeLists.txt:15 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done (0.7s)
-- Generating done (0.3s)
-- Build files have been written to: C:/Users/gageg/Desktop/force-riscv/build
PS C:\Users\gageg\Desktop\force-riscv\build> mingw32-make.exe
[  1%] Building CXX object riscv/CMakeFiles/friscv.dir/src/AddressFilteringRegulatorRISCV.cc.obj
[  1%] Building CXX object riscv/CMakeFiles/friscv.dir/src/AddressTableManagerRISCV.cc.obj
[  1%] Building CXX object riscv/CMakeFiles/friscv.dir/src/ArchInfoRISCV.cc.obj
[  2%] Building CXX object riscv/CMakeFiles/friscv.dir/src/BootOrderRISCV.cc.obj
[  2%] Building CXX object riscv/CMakeFiles/friscv.dir/src/EnumsRISCV.cc.obj
[  3%] Building CXX object riscv/CMakeFiles/friscv.dir/src/GenExceptionAgentRISCV.cc.obj
[  3%] Building CXX object riscv/CMakeFiles/friscv.dir/src/GenSequenceAgentRISCV.cc.obj
[  4%] Building CXX object riscv/CMakeFiles/friscv.dir/src/GeneratorRISCV.cc.obj
[  4%] Building CXX object riscv/CMakeFiles/friscv.dir/src/InitialSetupRISCV.cc.obj
[  4%] Building CXX object riscv/CMakeFiles/friscv.dir/src/InstructionRISCV.cc.obj
[  5%] Building CXX object riscv/CMakeFiles/friscv.dir/src/MemoryTraitsRISCV.cc.obj
[  5%] Building CXX object riscv/CMakeFiles/friscv.dir/src/OperandConstraintRISCV.cc.obj
[  6%] Building CXX object riscv/CMakeFiles/friscv.dir/src/OperandRISCV.cc.obj
[  6%] Building CXX object riscv/CMakeFiles/friscv.dir/src/PageRequestRegulatorRISCV.cc.obj
[  7%] Building CXX object riscv/CMakeFiles/friscv.dir/src/PhysicalPageManagerRISCV.cc.obj
[  7%] Building CXX object riscv/CMakeFiles/friscv.dir/src/PteAttributeRISCV.cc.obj
[  7%] Building CXX object riscv/CMakeFiles/friscv.dir/src/PyEnvironmentRISCV.cc.obj
In file included from D:/VScode/mingw64/lib/gcc/x86_64-w64-mingw32/8.1.0/include/c++/math.h:36,
                 from D:/VScode/Python3.6/include/pyport.h:194,
                 from D:/VScode/Python3.6/include/Python.h:53,
                 from C:/Users/gageg/Desktop/force-riscv/3rd_party/inc/pybind11/detail/common.h:117,
                 from C:/Users/gageg/Desktop/force-riscv/3rd_party/inc/pybind11/pytypes.h:17,
                 from C:/Users/gageg/Desktop/force-riscv/3rd_party/inc/pybind11/cast.h:18,
                 from C:/Users/gageg/Desktop/force-riscv/3rd_party/inc/pybind11/attr.h:18,
                 from C:/Users/gageg/Desktop/force-riscv/3rd_party/inc/pybind11/pybind11.h:44,
                 from C:/Users/gageg/Desktop/force-riscv/3rd_party/inc/pybind11/embed.h:17,
                 from C:\Users\gageg\Desktop\force-riscv\riscv\src\PyEnvironmentRISCV.cc:18:
D:/VScode/mingw64/lib/gcc/x86_64-w64-mingw32/8.1.0/include/c++/cmath:1121:11: error: '::hypot' has not been declared
   using ::hypot;
           ^~~~~
mingw32-make[2]: *** [riscv\CMakeFiles\friscv.dir\build.make:316: riscv/CMakeFiles/friscv.dir/src/PyEnvironmentRISCV.cc.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:991: riscv/CMakeFiles/friscv.dir/all] Error 2
mingw32-make: *** [Makefile:165: all] Error 2

gcc-hub, 2023-09-04:
> It seems u build it on windows. We didn't test force-riscv on Windows before.

yes! For others who have experience using it on windows, they may have encountered similar issues.

hanfeng0114, 2023-09-06:
It is not expected to work on Windows.
Currently, linux is the only choice. Will support macos later.
Try docker flow as what the wiki said: https://github.com/openhwgroup/force-riscv/wiki/Build-Environment

closed 2023-09-15 by gcc-hub