openhwfoundation/force-riscv #82: Build Failed issues

Sherleo, 2023-08-29:
gcc version : gcc/9.2.0

export FORCE_CC=/usr/bin/g++
export FORCE_PYTHON_VER=3.8
export FORCE_PYTHON_LIB=/tools/misc/anaconda3-2020/lib
export FORCE_PYTHON_INC=/tools/misc/anaconda3-2020/include/python3.8
make
Getting following error

/tools/misc/gcc/9.2.0/bin/g++ -c -Wall -std=c++11 -gdwarf-3 -m64  -O2 -Weffc++  -D 'ARCH_ENUM_HEADER="Enums.h"' -DUNIT_TEST -I./inc -I../base/inc -I../3rd_party/inc -I../../utils/handcar -I/tools/misc/anaconda3-2020/include/python3.8  -o make_area/obj/load_program_options.o ../src/load_program_options.cc
../src/load_program_options.cc: In function 'void Force::load_program_options(Force::ConfigFPIX&, std::vector<std::__cxx11::basic_string<char> >&, std::vector<std::__cxx11::basic_string<char> >&, int , char**,const string&)':
../src/load_program_options.cc:161:20: error: ‘fpixUsage’ was not declared in thsi scope
     161 |                std::cout << fpixUsage << std::endl;
            |                                    ^~~~~~~

Sherleo, 2023-08-29:
@hanfeng0114 Hi, can you help me about this issue?

Sherleo, 2023-08-29:
This issue is caused by the fact that the python script did not execute successfully and did not generate usage.h

hanfeng0114, 2023-08-29:
Please refer to #76 and the wiki https://github.com/openhwgroup/force-riscv/wiki first.
If not solved, reply the issue again.

Sherleo, 2023-08-29:
> Please refer to #76 and the wiki https://github.com/openhwgroup/force-riscv/wiki first. If not solved, reply the issue again.

This issue has been solved ， but there are some new compile error occurs when link object files to fpix_riscv .

main.cc:53 Force::load_program_options(...) , main.cc:92 Force::PluginManager(...) , undefined reference.

Sherleo, 2023-08-30:
This is a issue when compile on cetos7，but can work on ubuntu

closed 2023-08-30 by Sherleo