openhwfoundation/force-riscv #86: I need help about FlatMap

kong-ling-hui, 2023-10-11:
[image]
I excute this case, where the FlatMap is 1 or 0. But the VA always equals the PA.
[image]

hanfeng0114, 2023-10-12:
Do u mean FlatMap=1/0 in the _def_ctrl.py file or genVAforPA parameters?
From your context, I assume it is _def_ctrl.py option.
Please note that RISCV M mode is always direct map VA/PA (force-riscv doesn't support mstatus.MPRV=1 right now).
So if u want random virtual memory mapping, please change mode first.

hanfeng0114, 2023-10-16:
Is there any more question about this issue?

kong-ling-hui, 2023-10-17:
Thanks for your answer. I have switched to "S" mode by using the instruction "MRET", but I want to know did u have other methods to switch the mode. Looking forward to your guidance.

hanfeng0114, 2023-10-17:
Please refer to tests/riscv/privilege_switch for demo test to switch priviledge.

hanfeng0114, 2023-10-17:
Please reopen the issue if there is more question about this issue.

closed 2023-10-17 by hanfeng0114

xusq123, 2024-09-06:
> ![image](https://private-user-images.githubusercontent.com/63539162/274189087-dda886ee-852f-42b7-af64-857b16f6e45c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

how to print the(Mem R......) all my log don't have this