openhwfoundation/force-riscv #88: After MRET，not the PC

kong-ling-hui, 2023-10-20:
I used the following Python script to switch the privilege level to S or U mode, and then inserted the instructions to be executed after MRET. I found that the PC segment is not in 80011XXX. I have modified the PC in the boot section and the initial PC in PcConfig, hoping that the inserted PC will start at 80011000

kong-ling-hui, 2023-10-20:
here is the Python script,

kong-ling-hui, 2023-10-20:
[image]
the following is the MERT instruction I insert, and as the picture show the privilege level has been switched to S mode.
[image]
But the PC of instructions after MRET is not 80011XXX
[image]

hanfeng0114, 2023-10-20:
Force will try to random new instruction pc space after return executing xret instruction, which will set new pc for xepc register.
So it is not expected to execute from xret pc+4.
If u want to constraint pc area, try to modify memory.py.

kong-ling-hui, 2023-10-20:
[image]
I modified the address range, but when generating the case, it looked like it was stuck
[image]
And could I modify the xepc register to execute from specified address?

kong-ling-hui, 2023-10-20:
And another question, how to know "genInstruction" have this param, I didn't find "priv" or "epc" in FORCE-RISCV_User_Manual-v0.8.
[image]

kong-ling-hui, 2023-10-20:
> [image] I modified the address range, but when generating the case, it looked like it was stuck [image] And could I modify the xepc register to execute from specified address?

I know the reason, that the VA generated didn't match range I specified, so it is generate new VA. And could I modify the xepc register to execute from specified address?

hanfeng0114, 2023-10-21:
Note it is not expected to include all instruction/data areas (except device areas) in memory.py. genVA/genPA/PC will randomize addresses based on the settings.

hanfeng0114, 2023-10-21:
"epc" "priv" are not common parameters for a series of instructions. So it is not detailed in the spec.
Just follow the example.

hanfeng0114, 2023-10-21:
Please reopen the issue if u have more questions.

closed 2023-10-21 by hanfeng0114

kong-ling-hui, 2023-10-24:
There are currently no issues. I appreciate your support very much.