openhwfoundation/force-riscv #89: After 'MRET', generating Privilege mode randomly

kong-ling-hui, 2023-10-25:
- [ 1 ] Why randomly, the Privilege will be the value in MPP according to SPEC. 

- [ 2 ] I wrote the MPP to U and used {reseveRegister(name='mstatus', access='Write')}, but force always write the 'mstatus' randomly

[image]
[image]

kong-ling-hui, 2023-10-25:
I know how to switch specific Privilege using {genInstrution(MRET,{privXXX})}, but I want to try other methods, as mentioned above.

hanfeng0114, 2023-10-30:
1. a default preamble sequence will be added before the expected "MRET" instruction is generated by the force-riscv backend
2. if user has predefined sequence class, try to modify the following line in the variable_rvxx.xml:
  <variable name="Ret Preamble Sequence Class" value="" type="String" />

What's more, reserve register means only the reserved resource will not be randomly chosen.

So try to use "mret" sequence with parameters or use the priviledge switch functions or even state transition related APIs.

hanfeng0114, 2023-10-31:
Close the issue.
if more question, please reopen or issue new issues.

closed 2023-10-31 by hanfeng0114