openhwfoundation/force-riscv #92: where i can find the code for bootloading,in genagentseq

xusq123, 2024-02-07:
but  i can find how the instruction generating
also branch not taken

thanks

hanfeng0114, 2024-02-19:
1.U can check InitialSetupRISCV.cc for basic register initialization. U can also add your own GenThreadInitialization function as "tests/riscv/exception/trap_vm_force.py" does. Force framework will make boot loading based on register initial.
2.please refer to tsts/riscv/bnt for examples.

xusq123, 2024-02-23:
1:tests/riscv/exception/trap_vm_force.py" this case run fail
2:我按照玄铁的操作方式，将.s文件转换成了.pat，也能在c910上加载波形，看到相应的PC值；加上了波形停止的条件。这是因为.obj有详细的介绍每段代码是干什么的；但是force_riscv生成的.ELF文件转换位。obj是多个text段；text0是threadsplit,text1是initsetup,text2是mainseq，这些都很好理解，且相应的代码也能找到，但是，因为bootloading的PC跟initsetup的地址是一样的，这就不知道bootloading是那一段text，或者说bootloading的代码无法找到；而且有段text会随机产生一个PC，然后去align,这个过程就找不到代码，这个是封装好了的吗

MikeOpenHWGroup, 2024-02-23:
Hi @xusq123, may I ask you to use English for OpenHW Group GitHub discussions, issues and pull-request?  We are a global community and use English for all public communication. 谢谢您的大力支持!

hanfeng0114, 2024-02-28:
There are lots of work to do if u want to integrate force-riscv to one target platform, like memory map, riscv extension version support .etc.
It seems u want to integrate force-riscv into openxuantie c910 project:
1.please make sure the memory map first. check the former issue for how to change force-riscv memory map
2.force-riscv's section/segments are meaningless. actually each section/segment is only some continuous memory region.
3.bootloading stage is a scheme force-riscv deployed to initial registers, which is template directed random generated

hanfeng0114, 2024-02-28:
Please join mattermost for open discussion. Close this issue first.

closed 2024-02-28 by hanfeng0114