openhwfoundation/force-riscv #99: Instruction vmandnot and vmornot have been changed to vmandn and vmorn in Vector Extension Version 1.0

hcube123, 2024-05-21:
In RISC-V "V" Vector Extension Version 1.0 chapter 15.1, instruction `vmandnot` and `vmornot` have been changed to `vmandn` and `vmorn`.
But in the `riscv/arch_data/v_instructions.xml`, the instructions are still the old version.

Are there any differences in the handcar implementation between the old and new versions, Or should I just change this xml file?

hanfeng0114, 2024-06-20:
The current version only support V0.9 version on force-riscv side.
I have issue an pr which integrate V1.0 spike. And u can have a try with it, which should only update force-riscv side codes.
It is really hard for me to spare time to opensource our own code.
Have a try if u want to update.

hcube123, 2024-06-20:
您好，您的邮件已收到，我将尽快给您答复，谢谢O(∩_∩)O！