openhwfoundation/force-riscv #100: H extension for VS privilege

kong-ling-hui, 2024-06-28:
I have added VS privilege in PrivilegeLevel.py and some H CSR initial program. Now I can pass compilation, but when I generate case, which just has a simple "ld/sd" instr, I encounter some errors. I want to know the reason of this errors. Can you give me some help?
`[info]{GenQueryAgent::GenState} state: PC=0x80eb03a4
[notice]Front-end: INTERNAL ERROR (EnvironmentCallHandlerRISCV) BRANCH TARGET (LOAD_FROM_DATA_BLOCK) MISMATCH:
[notice]Front-end: INTERNAL ERROR (EnvironmentCallHandlerRISCV) BRANCH ADDRESS: 0x80eb0318, BRANCH OFFSET (# OF INSTRS): 48, LABEL OFFSET: 70
[fail]Front-end error reported: INTERNAL ERROR (EnvironmentCallHandlerRISCV) LABEL ADDRESS: 0x80eb03a4, LABEL OFFSET (# OF INSTRS): 70
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/Env.py", line 145, in generate
    self.mGenMain.setup()
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/GenThread.py", line 142, in setup
    seq.run()
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/Sequence.py", line 49, in run
    self.generate(**kargs)
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/GenThread.py", line 38, in generate
    self.setupEssentials()
  File "/nfs/home/konglinghui/tools/force-riscv/py/riscv/GenThreadRISCV.py", line 60, in setupEssentials
    self.genThread.installHandlers()
  File "/nfs/home/konglinghui/tools/force-riscv/py/riscv/GenThreadRISCV.py", line 126, in installHandlers
    self.exceptionHandlerManager.run(address_table_manager=self.addressTableManager)
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/Sequence.py", line 49, in run
    self.generate(**kargs)
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/exception_handlers/ExceptionHandlerManager.py", line 146, in generate
    default_set_name=self.default_set_name,
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/exception_handlers/ThreadHandlerSet.py", line 115, in generate
    self._genExcepHandlerCombos()
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/exception_handlers/ThreadHandlerSet.py", line 233, in _genExcepHandlerCombos
    self._genPrivilegeLevelSecurityStateHandlerSet(priv_level, security_state)
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/exception_handlers/ThreadHandlerSet.py", line 285, in _genPrivilegeLevelSecurityStateHandlerSet
    sync_dispatch_addr = handler_set.generateSynchronousHandlers(securityState, dispatcher)
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/exception_handlers/PrivilegeLevelHandlerSet.py", line 131, in generateSynchronousHandlers
    aSyncDispatcher,
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/exception_handlers/PrivilegeLevelHandlerSet.py", line 273, in _generateSynchronousDispatchLevel
    aSecurityState,
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/exception_handlers/PrivilegeLevelHandlerSet.py", line 302, in _generateSynchronousHandler
    self._generateSynchronousHandlerRoutine(aMemBank, aSecurityState, err_code, handler)
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/exception_handlers/PrivilegeLevelHandlerSet.py", line 349, in _generateSynchronousHandlerRoutine
    handler_context=self.createExceptionHandlerContext(aErrCode, aMemBank),
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/exception_handlers/ReusableSequence.py", line 48, in generateRoutine
    self._generateValidatedRoutine(routine_name, **kwargs)
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/exception_handlers/ReusableSequence.py", line 120, in _generateValidatedRoutine
    routine_gen_method(**kwargs)
  File "/nfs/home/konglinghui/tools/force-riscv/py/riscv/exception_handlers/EnvironmentCallHandler.py", line 83, in generateHandler
    self.mAssemblyHelper.addLabel("LOAD_FROM_DATA_BLOCK")
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/AssemblyHelper.py", line 290, in addLabel
    "(# OF INSTRS): %d" % (self._mLabelRoutineName, cur_pc, label_offset)
  File "/nfs/home/konglinghui/tools/force-riscv/py/base/Sequence.py", line 363, in error
    stack_frame_str = get_stack_frame_string()

[FAIL]{front-end-error-reported} in file '../base/inc/py_modules/PyLog.h' line 58 func 'operator()'.`

kong-ling-hui, 2024-07-04:
The reason of this error is that the destination of instruction "JAL" is not meeting FORCE expectation. I have skip this error-detected mechanism about this error and FORCE normally generate case, but this error is still not completely solved. Looking forward to your reply~

xusq123, 2024-09-06:
what the role of the AddressTable.CC and its Manager.CC