openhwfoundation/force-riscv #101: DIFFERENCE between EVmRegimeType and EPrivilegeLevelType?

kong-ling-hui, 2024-07-04:
if I want to make FORCE supporting H extension, should I need to modify EPrivilegeLevelType to H, or other value?
[image]

kong-ling-hui, 2024-07-17:
I will set regimeType according to priv.

closed 2024-07-17 by kong-ling-hui

xusq123, 2024-08-22:
how to add hypersivor in force_riscv   have you successful

kong-ling-hui, 2024-08-30:
> how to add hypersivor in force_riscv have you successful

emm… It will waste you a lot of time, I recommend you find a ISG already support H extension.

xusq123, 2024-09-02:
Is there a code that reserves support for virtual machines in force_riscv ?or  it may be zero to do

kong-ling-hui, 2024-09-02:
> Is there a code that reserves support for virtual machines in force_riscv ?or it may be zero to do

I think "Vm" in "EVmRegimeType" means virtual memory.

xusq123, 2024-09-03:
> > Is there a code that reserves support for virtual machines in force_riscv ?or it may be zero to do
> 
> I think "Vm" in "EVmRegimeType" means virtual memory.

so it mean force_riscv not have virtual machine.just have virtual memory

xusq123, 2024-10-11:
> > Is there a code that reserves support for virtual machines in force_riscv ?or it may be zero to do
> 
> I think "Vm" in "EVmRegimeType" means virtual memory.

Do you know the algorithm for page table conversion?，how to PTE work